openbmc/openbmc #724: Romulus warnings at boot

shenki, 2016-11-01:
Received a journal log from @mine260309 [journal.txt](https://github.com/openbmc/openbmc/files/563458/journal.txt)

### ipmid
```
Oct 24 02:22:05 romulus ipmid[673]: Failed to get /org/openbmc/settings/host0 busname: No route to host
```

### LED manager
LED manager needs to stop restarting in the case were there are no LEDs
```
Oct 24 02:22:12 romulus led_controller.exe[684]: No LEDs present in the system
Oct 24 02:22:12 romulus systemd[1]: Started Phosphor LED Management Daemon.
Oct 24 02:22:12 romulus systemd[1]: org.openbmc.control.led.service: Service hold-off time over, scheduling restart.
Oct 24 02:22:12 romulus systemd[1]: Stopped Phosphor LED Management Daemon.
Oct 24 02:22:12 romulus systemd[1]: Starting Phosphor LED Management Daemon...
Oct 24 02:22:13 romulus led_controller.exe[686]: No LEDs present in the system
Oct 24 02:22:13 romulus systemd[1]: Started Phosphor LED Management Daemon.
Oct 24 02:22:14 romulus systemd[1]: org.openbmc.control.led.service: Service hold-off time over, scheduling restart.
Oct 24 02:22:14 romulus systemd[1]: Stopped Phosphor LED Management Daemon.
Oct 24 02:22:14 romulus systemd[1]: org.openbmc.control.led.service: Start request repeated too quickly.
```

### Mapper
```
Oct 24 02:22:33 romulus systemd[1]: Started Initialize BIOS Flash.
Oct 24 02:22:33 romulus mapper[716]: Error finding service: No such file or directory
Oct 24 02:22:33 romulus systemd[1]: obmc-flash-init.service: Main process exited, code=exited, status=1/FAILURE
Oct 24 02:22:33 romulus systemd[1]: obmc-flash-init.service: Unit entered failed state.
Oct 24 02:22:33 romulus systemd[1]: obmc-flash-init.service: Failed with result 'exit-code'.
```

### GPIO devicetree/userspace conflict?
```
Oct 24 02:22:39 romulus button_power.exe[719]: GPIO Lookup:  POWER_BUTTON = 387,in
Oct 24 02:22:39 romulus kernel: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin C16 already requested by 1e630000.flash-controller; cannot claim for 1e780000.gpio:387
Oct 24 02:22:39 romulus kernel: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-67 (1e780000.gpio:387) status -22
Oct 24 02:22:39 romulus kernel[639]: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin C16 already requested by 1e630000.flash-controller; cannot claim for 1e780000.gpio:387
Oct 24 02:22:39 romulus kernel[639]: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-67 (1e780000.gpio:387) status -22
```

mine260309, 2016-11-01:
Regarding GPIO conflict, it's found that the GPIOI3's pin defined in `drivers/pinctrl/aspeed/pinctrl-aspeed-g5.c`:
```
  #define C16 67
  SIG_EXPR_DECL(SYSMISO, SPI1DEBUG, COND1, SPI1DEBUG_DESC);
  SIG_EXPR_DECL(SYSMISO, SPI1PASSTHRU, COND1, SPI1PASSTHRU_DESC);
  SIG_EXPR_LIST_DECL_DUAL(SYSMISO, SPI1DEBUG, SPI1PASSTHRU);
  SS_PIN_DECL(C16, GPIOI3, SYSMISO);
...
  FUNC_GROUP_DECL(SPI1DEBUG, C18, E15, B16, C16, B15, C15, A14, A15);
  FUNC_GROUP_DECL(SPI1PASSTHRU, C18, E15, B16, C16, B15, C15, A14, A15);
```
So it looks like `GPIOI3` is expected to be part of `SPI1DEBUG` and `SPI1PASSTHRU`, however, this GPIO is used as POWER_BUTTON according to schematic, and it applies to both Romulus and Witherspoon. (I guess Witherspoon should have the same issue?)

I'm not sure if this is the root cause, @shenki could you help to check this?

amboar, 2016-11-01:
@mine260309 Given that Romulus is strapped in spi1debug mode this will certainly cause issues. To the best of my knowledge the groups are accurate, so the strapping needs to be changed. Debug mode is marked as reserved in the datasheet - do we actually know what we're doing by enabling it? Does anyone have information on (presumably) snooping via the ahb bridge?

shenki, 2016-11-02:
It has been identified that the Witherspoon and Romulus strapping for SPI1 is incorrect. We've let the hardware team know.

I will cook up a patch to modify the strapping in the board file for now.

shenki, 2016-11-02:
Patches posted for review:

http://patchwork.ozlabs.org/patch/690261/
http://patchwork.ozlabs.org/patch/690260/

mine260309, 2016-11-02:
I confirm that @shenki's patches work, now the GPIO conflict issue is gone.
And we get other GPIO errors though.
Attached [journal.txt](https://github.com/openbmc/openbmc/files/567181/journal.txt)

mine260309, 2016-11-02:
And it's noticed that sometime `flasher.exe` process takes 100% CPU:
```
  719   703 root     R      97m  20% 100% flasher.exe bios  /org/openbmc/control/flash/bios
```

amboar, 2016-11-03:
@mine260309 #713 is probably related.

adamliyi, 2016-11-03:
Opened a related issue openbmc/openbmc#743 for Error message:
```
Oct 24 02:22:42 romulus system_manager.py[692]: ERROR: BMC_THROTTLE not found in GPIO config table
Oct 24 02:22:42 romulus system_manager.py[692]: ERROR: IDBTN not found in GPIO config table
```

mine260309, 2017-02-27:
This issue is not valid anymore since we have many changes on OpenBMC and Romulus.
@williamspatrick could you help to close this issue?

closed 2017-02-27 by shenki