openbmc/openbmc #734: Support the callout of a logical entity within an error log

geissonator, 2016-11-01:


geissonator, 2016-11-29:
We probably should break this into 2 stories.  1 where we figure out the design and define the yaml to do the callouts and update the python code to handle them, then another where we write the code to implement the callouts in the elog server.

dkodihal, 2016-12-01:
@geissonator : how about getting the requirements and design (put into YAML) with the first story?

dkodihal, 2016-12-01:
- [ ] Requirements : what should be contained in a callout section.
- [ ] Design : Specify a callout section in errorlog YAML.
- [ ] Design : Specify errorlog server behavior upon identifying a callout.

geissonator, 2016-12-02:
Opened up #854 to track implementation.

geissonator, 2016-12-14:
Deepak out next sprint so just putting in actual, not committed so we don't lose track of it.

dkodihal, 2017-01-12:
Code up for review @ https://gerrit.openbmc-project.xyz/#/c/1752

closed 2017-01-19 by geissonator