openbmc/openbmc #738: OpenBMC- Factory reset & zero config through IPMI is not working

prkatti1, 2016-11-02:
Factory reset & zero config through IPMI is not working.
We used following IPMI commands to reset to factory
> ipmitool -H $system_IP -I lanplus -U ADMIN -P admin raw 0x32 0xBA 00 00
> ipmitool -H $system_IP -I lanplus -U ADMIN -P admin raw 0x32 0x66

Commands are running successful, but without any impact on system.

williamspatrick, 2016-11-02:
We do not support any ipmi interface using the lanplus access method today. We do not plan on supporting all of the OEM commands that the AMI implementation on some open-power systems supported. What is it you are expecting to work?

sivassrr, 2016-11-02:
@williamspatrick Is there a way to do MFG reset . We want to test zeroconf, Patrick.

williamspatrick, 2016-11-02:
There is not a REST command yet to return the machine to manufacturer new state. That will be tackled as part of code update work. 

I thought when you do a code update now you identify which pieces to preserve. Anytime you do a full image rewrite (including the rw partition) without preserving anything it should go back to a clean state, doesn't it?

sivassrr, 2016-11-02:
Yes @williamspatrick we can try that, but that may not be same as factory reset,  I guess. We expect some default IP address. We can try without preserving IP address during code update.

prkatti1, 2016-11-07:
We used dbus commands here to do factory reset

geissonator, 2016-12-08:
Opened #865 to track this function, it will be 2Q2017 before this is ready.

closed 2016-12-08 by geissonator