openbmc/openbmc #739: [Witherspoon VERSION_ID="v1.99.0-212-gf1c10dd-dirty" ] code update succeed but fails with REST Authentication

sivassrr, 2016-11-02:
Had code update on witherspoon with 329 build. 
cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-212"
VERSION_ID="v1.99.0-212-gf1c10dd-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-212"
BUILD_ID="v1.99.0"
root@witherspoon:~# obmcutil state
 = BMC_READY

Able to ping, Code update succeed but REST Authentication failed with new level. 

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://<IP of Witherspoon>/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
curl: (28) Operation timed out after 0 milliseconds with 0 out of 0 bytes received

sivassrr, 2016-11-02:
Initiate Code update BMC :: BMC code update process initiation        | FAIL |
Keyword 'Verify Ping and REST Authentication' failed after retrying for 30 minutes. The last error was: REST Authentication Failed
------------------------------------------------------------------------------
Update Bmc :: Trigger code update to a target BMC.                    | FAIL |
1 critical test, 0 passed, 1 failed
1 test total, 0 passed, 1 failed

sivassrr, 2016-11-02:
[journaldumpcodeupdate.txt](https://github.com/openbmc/openbmc/files/565810/journaldumpcodeupdate.txt)

williamspatrick, 2016-11-02:
```gevent[856]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 42, in <module>
Oct 20 20:04:21 witherspoon phosphor-gevent[856]:     from tempfile import TemporaryFile
Oct 20 20:04:21 witherspoon phosphor-gevent[856]: ImportError: cannot import name TemporaryFile
Oct 20 20:04:21 witherspoon systemd[1]: phosphor-gevent.service: Main process exited, code=exited, status=1/FAILURE```

sivassrr, 2016-11-02:
Mem: 161004K used, 338004K free, 49364K shrd, 9768K buff, 80264K cached
CPU: 100% usr   0% sys   0% nic   0% idle   0% io   0% irq   0% sirq
Load average: 6.74 6.76 6.85 8/101 3948
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  806   795 root     R      97m  20% 100% flasher.exe bios  /org/openbmc/control
  819     1 root     S    32636   7%   0% /usr/sbin/power_control.exe 3000 10
  818     1 root     S    32632   7%   0% /usr/sbin/control_host.exe
  821     1 root     S    32628   7%   0% /usr/sbin/button_power.exe
  795     1 root     S    32520   7%   0% /usr/sbin/flash_bios.exe
  746     1 root     S    32516   7%   0% /usr/sbin/host_checkstop.exe
  753     1 root     S    32516   7%   0% /usr/sbin/host_watchdog.exe
  754     1 root     S    32516   7%   0% /usr/sbin/control_bmc.exe
  820     1 root     S    32516   7%   0% /usr/sbin/button_reset.exe
    1     0 root     R    21860   4%   0% {systemd} /sbin/init
  739     1 root     S    17904   4%   0% python /usr/sbin/netman.py
  732     1 root     S    16356   3%   0% python /usr/sbin/obmc-phosphor-userd
  796     1 root     S    15980   3%   0% python /usr/sbin/bmc_update.py
  744     1 root     S    15892   3%   0% python /usr/sbin/phosphor-mapper
  738     1 root     S    15652   3%   0% python /usr/sbin/inventory_items.py
  807     1 root     R    15580   3%   0% python /usr/sbin/hwmon.py
  783     1 root     S    15580   3%   0% python /usr/sbin/system_manager.py
  743     1 root     S    15572   3%   0% python /usr/sbin/sensor_manager2.py
  740     1 root     S    15540   3%   0% python /usr/sbin/download_manager.py
  755     1 root     S    15412   3%   0% {settings_manage} /usr/bin/python -u /
  731     1 root     S    15340   3%   0% python /usr/sbin/chassis_control.py

sivassrr, 2016-11-04:
As of now, I found REST works using the latest build for witherspoon - which I had flashed using pflash. https://openpower.xyz/job/openbmc-build/332/distro=ubuntu,target=witherspoon/artifact/images/witherspoon/

williamspatrick, 2016-11-04:
This sounds like you are saying this is a duplicate of the flash images issues?  Can you close if that is the case?

sivassrr, 2016-11-07:
@williamspatrick Still I am seeing failure in witherspoon - Related to REST.

sivassrr, 2016-11-07:
With Nov 4 level, Observed that 32 TCs associate with REST fails out of 50 
50 test total, 18 passed, 32 failed.  
=================================================================
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-216"
VERSION_ID="v1.99.0-216-gcecba5f-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-216"
BUILD_ID="v1.99.0"
root@witherspoon:~#

gkeishin, 2016-11-07:
On the following driver VERSION_ID="v1.99.0-216-gcecba5f-dirty"
root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~#

**A basic REST POST operation to get the system state is throwing error, example**

Oct 20 21:03:50 witherspoon phosphor-gevent[753]: ::c0f3:a376:8e91:f402%1260622248 - - [2016-10-20 21:03:50] "POST /login HTTP/1.1" 200 320 0.031019
Oct 20 21:03:50 witherspoon phosphor-gevent[753]: Traceback (most recent call last):
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 858, in _handle
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     route, args = self.router.match(environ)
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 740, in custom_router_match
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     route.callback._setup(**args)
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 80, in _setup
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     return self.setup(**kw)
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 203, in setup
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     request.route_data['method'] = self.find(path, method)
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 194, in find
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     self.mapper.get_object, path=path)
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 97, in try_mapper_call
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     return f(**kw)
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 56, in get_object
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     retries)
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 42, in retry
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     return func()
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 55, in <lambda>
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     path, interfaces, signature='sas'),
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     **keywords)
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Oct 20 21:03:50 witherspoon phosphor-gevent[753]:     message, timeout)
Oct 20 21:03:50 witherspoon phosphor-gevent[753]: DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.13 was not provided by any .service files
Oct 20 21:03:50 witherspoon phosphor-gevent[753]: ::c0f3:a376:8e91:f402%1260622248 - - ```[2016-10-20 21:03:50] "POST /org/openbmc/managers/System//action/getSystemState HTTP/1.1" 500 2095 0.062594 ```

sivassrr, 2017-05-25:
No longer observe this. Closing it.

closed 2017-05-25 by sivassrr