openbmc/openbmc #740: [Palmetto - v1.99.0-212-gf1c10dd] - Bios update fails on Palmetto with both REST and pflash

rahulmah, 2016-11-02:
Bios update on Palmetto is failing with both REST and pflash method. I have used Palmetto bios build #205 (md5sum e3bb9045efe8a7681c037d563b92fc50) and #199(md5sum 5b44316a31d1fc0e8498e84e7136a26f). 

With REST, Bios update command works fine. But just after this command, flash status went into "Flash Error" state. 

```
Test Bios Update :: This testsuite updates the PNOR image on the host for
==============================================================================
Host BIOS Update And Boot :: This test updates the PNOR image on t... URI:https://xx.xx.xx.xx/org/openbmc/control/chassis0//action/powerOff, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
URI:https://xx.xx.xx.xx/org/openbmc/control/chassis0//action/getPowerState, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": 0,
  "message": "200 OK",
  "status": "ok"
}
URI:https://xx.xx.xx.xx/org/openbmc/records/events//action/clear, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": 0,
  "message": "200 OK",
  "status": "ok"
}
.URI:https://xx.xx.xx.xx/org/openbmc/control/flash/bios//action/update, method:Post, args:{u'data': {u'data': [u'/tmp/palmetto205.pnor']}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
URI:https://xx.xx.xx.xx/org/openbmc/control/flash/bios, method:Get, args:{}
Response code:200, Content:{
  "data": {
    "filename": "/tmp/palmetto205.pnor",
    "flasher_instance": "bios",
    "flasher_name": "flasher.exe",
    "flasher_path": "/usr/sbin/flasher.exe",
    "lock": 0,
    "name": "",
    "status": "Flash Error"       <---- Flash Error seen just after triggering update.
  },
  "message": "200 OK",
  "status": "ok"
}
Host BIOS Update And Boot :: This test updates the PNOR image on t... | FAIL |
Keyword 'Is PNOR Flashing' failed after retrying for 2 minutes. The last error was: Flash Error != Flashing
------------------------------------------------------------------------------
Test Bios Update :: This testsuite updates the PNOR image on the h... | FAIL |
1 critical test, 0 passed, 1 failed
1 test total, 0 passed, 1 failed
==============================================================================
Output:  /gsa/ausgsa-p9/14/indiateam/rahulmah/OB_2Nov_run/output.xml
Log:     /gsa/ausgsa-p9/14/indiateam/rahulmah/OB_2Nov_run/log.html
Report:  /gsa/ausgsa-p9/14/indiateam/rahulmah/OB_2Nov_run/report.html
bash-4.1$
```

With pflash, bios update fails with "Verification failed for chunk at 0x00000000" error message.

```
root@palmetto:/tmp# pflash -p /tmp/palmetto.pnor -f -i -E
Flash info:
-----------
Name          = Macronix MXxxL51235F
Total size    = 64MB
Erase granule = 4KB

TOC@0x00000000 Partitions:
-----------
ID=00            part 00000000..00001000 (actual=00001000)
ID=01             HBI 00010000..005b0000 (actual=005a0000) [ECC]
ID=02            MVPD 005b0000..00640000 (actual=00090000) [ECC]
ID=03            CVPD 00640000..00688000 (actual=00048000) [ECC]
ID=04           DJVPD 00688000..006d0000 (actual=00048000) [ECC]
ID=05             HBD 006d0000..00730000 (actual=00060000) [ECC]
ID=06            SBEC 00730000..007c0000 (actual=00090000) [ECC]
ID=07             SBE 007c0000..00808000 (actual=00048000) [ECC]
ID=08        ATTR_TMP 00808000..00810000 (actual=00008000)
ID=09       ATTR_PERM 00810000..00818000 (actual=00008000) [ECC]
ID=10            WINK 00818000..00938000 (actual=00120000) [ECC]
ID=11           GUARD 00938000..0093d000 (actual=00005000) [ECC]
ID=12            HBEL 0093d000..00961000 (actual=00024000) [ECC]
ID=13         PAYLOAD 00961000..00a61000 (actual=00100000)
ID=14      BOOTKERNEL 00a61000..01961000 (actual=00f00000)
ID=15           NVRAM 01961000..019f1000 (actual=00090000)
ID=16            HBRT 019f1000..01d51000 (actual=00360000) [ECC]
ID=17             OCC 01d51000..01e71000 (actual=00120000) [ECC]
ID=18         FIRDATA 01e71000..01e74000 (actual=00003000) [ECC]
ID=19            CAPP 01e74000..01e98000 (actual=00024000) [ECC]
ID=20         SECBOOT 01e98000..01ebc000 (actual=00024000) [ECC]
ID=21         BMC_INV 01ebc000..01ec5000 (actual=00000000)
ID=22             HBB 01f67000..01ff7000 (actual=00090000) [ECC]
ID=23         VERSION 01ff7000..01ff8000 (actual=00001000)
ID=24     BACKUP_PART 01ff8000..02000000 (actual=00000000)
About to erase chip !
Erasing... (may take a while !) done !
About to program "/tmp/palmetto.pnor" at 0x00000000..0x02000000 !
Programming & Verifying...
[                                                  ] 0%LIBFLASH: Miscompare at 0x00000000
Verification failed for chunk at 0x00000000
root@palmetto:/tmp#
```

**BMC fw info** 
root@palmetto:/tmp/199# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-212"
VERSION_ID="v1.99.0-212-gf1c10dd-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-212"
BUILD_ID="v1.99.0"
root@palmetto:/tmp/199# cat /etc/version
20161102043616

shenki, 2016-11-02:
We moved to using mtd for flasher. This could be a regression caused by that.

I suggest testing with 6ddfdfb1ff120e855dbeb6c5a319e8d1b4dc3968 reverted to see if it helps. (The commit from skeleton we want to revert is https://github.com/openbmc/skeleton/commit/9d61237e4b669002b340fedcf66ce36f99ba32ce).

shenki, 2016-11-04:
Here are the patches that will resolve this:

https://gerrit.openbmc-project.xyz/#/q/flasher+is:open+openbmc/skeleton

Please review.

closed 2016-11-04 by williamspatrick