openbmc/openbmc #741: Witherspoon: Slow response with 1.99.0-212 driver

yadlapati, 2016-11-02:
Flashed new code on Witherspoon (w5.aus.stglabs.ibm.com).
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-212"
VERSION_ID="v1.99.0-212-gf1c10dd-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-212"
BUILD_ID="v1.99.0"

root@witherspoon:~# uname -a
Linux witherspoon 4.7.10-c5682cbc3542add705e35152329436067bb1f94a #1 Tue Nov 1 19:39:35 UTC 2016 armv6l GNU/Linux

System is very slow to respond. Top shows 100% CPU usage
Mem: 115168K used, 383840K free, 9084K shrd, 9392K buff, 39272K cached
CPU: 100% usr   0% sys   0% nic   0% idle   0% io   0% irq   0% sirq
Load average: 4.09 3.99 3.86 6/101 822
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  764   742 root     R      97m  20% 100% flasher.exe bios  /org/openbmc/control/flash/bios

dmesg shows:
aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin C16 already requested by 1e630000.flash-controller; cannot claim for 1e780000.gpio:387
aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-67 (1e780000.gpio:387) status -22
random: nonblocking pool is initialized

sivassrr, 2016-11-03:
https://github.com/openbmc/openbmc/issues/739 - May be REST failing because of flasher.exe issue, I guess.

geissonator, 2016-11-03:
Probably all intertwined with #740  There was a regression in the pflash area recently.

shenki, 2016-11-04:
Yes. This is a duplicate of #740.

closed 2016-11-04 by shenki

amboar, 2016-11-04:
Regarding

```
aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin C16 already requested by 1e630000.flash-controller; cannot claim for 1e780000.gpio:387
aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-67 (1e780000.gpio:387) status -22
random: nonblocking pool is initialized
```

This was a strapping issue that's resolved by http://patchwork.ozlabs.org/patch/691126/