openbmc/openbmc #742: TFTP code update fails with path

shenki, 2016-11-03:
I have my pnor on an image at `palm4/palmetto.pnor`.

```
curl -b cjar -k -H "Content-Type: application/json" -X POST     -d '{"data": ["10.61.2.7", "palm4/palmetto.pnor"]}'     https://10.61.254.187/org/openbmc/control/flash/bios/action/updateViaTftp
```

The update code struggles with this:

```
Nov 03 06:06:58 palmetto phosphor-gevent[782]: ::c063:abb6:8e91:f402%1230014792 - - [2016-11-03 06:06:58] "POST /org/openbmc/control/flash/bios/action/updateViaTftp HTTP/1.1" 401 230 0.215966
Nov 03 06:07:38 palmetto phosphor-gevent[782]: ::c063:abb6:8e91:f402%1230014792 - - [2016-11-03 06:07:38] "POST /login HTTP/1.1" 200 320 0.052616
Nov 03 06:07:57 palmetto download_manager.py[668]: Downloading: palm4/palmetto.pnor from 10.61.2.7
Nov 03 06:07:57 palmetto phosphor-gevent[782]: ::c063:abb6:8e91:f402%1230014792 - - [2016-11-03 06:07:57] "POST /org/openbmc/control/flash/bios/action/updateViaTftp HTTP/1.1" 200 169 0.261094
Nov 03 06:07:57 palmetto download_manager.py[668]: tftp: can't open '/tmp/palm4/palmetto.pnor': No such file or directory
```

williamli80, 2016-11-08:
@williamspatrick @geissonator  Who can help on this?

adamliyi, 2016-11-09:
@shenki , please refer to @mine260309 's patch here:
https://gerrit.openbmc-project.xyz/#/c/544/4/code-update.md

Looks to me the updateViaTftp only recognize *.tar.
```
Perform a POST to invoke the `updateViaTftp` method of the `/flash/bmc` object:

    curl -b cjar -k -H "Content-Type: application/json" -X POST \
        -d '{"data": ["<TFTP server IP address>", "<filename>"]}' \
        https://bmc/org/openbmc/control/flash/bmc/action/updateViaTftp

Note the `<filename>` shall be a tar-ball.
```

mine260309, 2016-11-09:
I met such issue before, the root cause is that the `<filename>` contains directory (here it's `palm4/palmetto.pnor`), the update script just simply does not handle such case, trying to save the file to `/tmp/palm4/palmetto.pnor`, where the parent directoy does not exist yet, that why it gets error `No such file or directory`.

We can address this issue by either:
* indicate in documentation that the filename shall not contains directory;

or:
* Modify the update script to handle the directory issue.

shenki, 2016-11-09:
I think we should modify the script to handle the directory issue. Once the file has been tftp'd over to the BMC, we should discard the directory part of the path.

mine260309, 2016-11-10:
Yup, I prefer to fix this issue as well.

ya-mouse, 2017-02-09:
Take a look at:
https://github.com/openbmc/skeleton/issues/133