openbmc/openbmc #743: Romulus error message: not found in GPIO config table

adamliyi, 2016-11-03:
From the Romulus boot log in: https://github.com/openbmc/openbmc/issues/724
```
Oct 24 02:22:42 romulus system_manager.py[692]: ERROR: BMC_THROTTLE not found in GPIO config table
Oct 24 02:22:42 romulus system_manager.py[692]: ERROR: IDBTN not found in GPIO config table
```
The error message will show up if there is no GPIO configure in system configure files (Romulus.py, Witherspoon.py).

In: https://github.com/openbmc/skeleton/blob/master/op-hostctl/control_host_obj.c#L280
the host boot code will invoke:
```
	gpio_init(connection,&fsi_data);
	gpio_init(connection,&fsi_clk);
	gpio_init(connection,&fsi_enable);
	gpio_init(connection,&cronus_sel);
	gpio_init(connection,&Throttle);
	gpio_init(connection,&idbtn);
```
But pins like `Throttle` or `idbtn` may not be necessary for Witherspoon/Romulus.

mine260309, 2016-11-03:
Regarding `IDBTN`, both Witherspoon and Romulus uses GPIOQ7, we can add this in config;
But for `BMC_THROTTLE`, it is not expected to be used in power on sequence (see Matt's comment in #648). So probably the related code in `control_host_obj.c` should be removed.

mine260309, 2016-11-03:
Part of the fix (GPIO config) is post for review: https://gerrit.openbmc-project.xyz/#/c/994/

mine260309, 2016-11-07:
The real fix should be to remove `IDBTN` and `BMC_THROTTLE` related code for Witherspoon and Romulus, but keep them for P8 machines.

mine260309, 2016-11-07:
From an email discussion, @williamspatrick mentioned that @shenki will have a refactor proposal on these parts of code, and currently we will focus on P9 power on.
So the temp fix could be define both `IDBTN` and `BMC_THROTTLE` for Witherspoon and Romulus, so that it does not block the power on sequence code.

adamliyi, 2016-11-08:
A better fix is mentioned by @GHF:
```
We had a similar issue in the op-pwrctl where the source hard-coded
the system net names. There I put in a short-term solution of adding a
function-specific abstraction over the net names into the System.py
config, removing the per-machine patches for name and polarity:
https://gerrit.openbmc-project.xyz/#/c/606/
https://gerrit.openbmc-project.xyz/#/c/689/

It's not difficult to extend POWER_CONFIG for GPIO functions that only
exist on one machine, like so:
https://gerrit.openbmc-project.xyz/#/c/608/3

With the abstraction taking place in the Python machine config, we
don't need per-machine dispatch in the user interface (op-pwrctl,
op-hostctl, op-pwr-button, etc).

Does that work for you in the short term?
```

adamliyi, 2016-11-08:
@mine260309 , you mentioned:
```
> In P9, BMC_THROTTLE is renamed to N_MODE_N, and changed to an IN GPIO;
> IDBTN is changed to IN GPIO as well.
> If we keep the above code and overriding the GPIOs, writing
> BMC_THROTTLE takes no effect, but writing to IDBTN may break the
> function of its related front panel button.
```
So define IDBTN in Witherspoon and Romulus will `break the function of its related front panel button`?

mine260309, 2016-11-08:
@adamliyi Yup, according to Nick's explanation, IDBTN on P9 is a IN GPIO connected with a front panel button, if user push the button, a LED in the back will light.
If BMC override and write this GPIO to 1, (from the schematic) it **may** cause the function to break, that pushing the button will not turn on the LED.

williamspatrick, 2016-11-18:
My understanding is this has been resolved.

closed 2016-11-18 by williamspatrick

mine260309, 2016-11-18:
Yes, the issue is resolved.
And I opened a new task to abstract GPIO for host-control #814, I will work on that task.