openbmc/openbmc #744: Need update in README from where new images are available (openpower.xyz)

sivassrr, 2016-11-04:
As REST is kind of broken due to flasher.exe https://github.com/openbmc/openbmc/issues/740 , 
So only way to move to new level is using pflash code update method only. 

It will be helpful to have details like which image to use for pflash and REST way. 
https://openpower.xyz/job/openbmc-build/332/distro=ubuntu,target=witherspoon/artifact/images/witherspoon/

Also if customer uses pflash then IP address will be gone. So need update on the same as well.

williamspatrick, 2016-11-04:
Why would we document a workaround for a bug?  On no open source project is master assumed to always be bug free. We are in the middle of transitioning from one driver to another which caused userspace impact as well. That is nearly all fixed, hopefully by Monday your time.

sivassrr, 2016-11-09:
@williamspatrick we are publishishing all the images https://openpower.xyz/job/openbmc-build/332/distro=ubuntu,target=witherspoon/artifact/images/witherspoon/,  Do we have document on which image should be used on which interface (Codeupdate via REST, Codeupdate via pflash). If it is available then we can close this.

sivassrr, 2017-05-25:
Release notes take care of it.  As of now, I am closing it.

closed 2017-05-25 by sivassrr