openbmc/openbmc #747: pflash fails on barreleye with latest e7b0fd5 (flasher and mtd fixes)

geissonator, 2016-11-04:
Nothing pops up in dmesg or journal on the fail.

```
cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-220"
VERSION_ID="v1.99.0-220-ge7b0fd5-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-220"
BUILD_ID="v1.99.0"

[barreleye]# pflash -E -f -i -p /tmp/barreleye.pnor
Flash info:
-----------
Name          = /dev/mtd7
Total size    = 64dMB 
Erase granule = 4KB 

TOC@0x00000000 Partitions:
-----------
ID=00            part 00000000..00001000 (actual=00001000) 
ID=01            HBEL 00008000..0002c000 (actual=00024000) [ECC]
ID=02           GUARD 0002c000..00031000 (actual=00005000) [ECC]
ID=03             HBD 00031000..00091000 (actual=00060000) [ECC]
ID=04           DJVPD 00091000..000d9000 (actual=00048000) [ECC]
ID=05            MVPD 000d9000..00169000 (actual=00090000) [ECC]
ID=06            CVPD 00169000..001b1000 (actual=00048000) [ECC]
ID=07             HBI 001b1000..00751000 (actual=005a0000) [ECC]
ID=08            SBEC 00751000..007e1000 (actual=00090000) [ECC]
ID=09             SBE 007e1000..00829000 (actual=00048000) [ECC]
ID=10            WINK 00829000..00949000 (actual=00120000) [ECC]
ID=11            HBRT 00949000..00ca9000 (actual=00360000) [ECC]
ID=12         PAYLOAD 00ca9000..00da9000 (actual=00100000) 
ID=13      BOOTKERNEL 00da9000..01ca9000 (actual=00f00000) 
ID=14        ATTR_TMP 01ca9000..01cb1000 (actual=00008000) 
ID=15       ATTR_PERM 01cb1000..01cb9000 (actual=00008000) [ECC]
ID=16             OCC 01cb9000..01dd9000 (actual=00120000) [ECC]
ID=17            TEST 01dd9000..01de2000 (actual=00009000) [ECC]
ID=18           NVRAM 01de2000..01e72000 (actual=00090000) 
ID=19         FIRDATA 01e72000..01e75000 (actual=00003000) [ECC]
ID=20         BMC_INV 01e75000..01e7e000 (actual=00000000) 
ID=21            CAPP 01e7e000..01ea2000 (actual=00024000) [ECC]
ID=22         SECBOOT 01ea2000..01ec6000 (actual=00024000) [ECC]
ID=23             HBB 01f60000..01ff0000 (actual=00090000) [ECC]
ID=24         VERSION 01ff7000..01ff8000 (actual=00001000) 
ID=25     BACKUP_PART 01ff8000..02000000 (actual=00000000) 
ID=26      OTHER_SIDE 02000000..02008000 (actual=00000000) 

TOC@0x02000000 Partitions:
-----------
ID=00            part 02000000..02001000 (actual=00001000) 
ID=01            HBEL 00008000..0002c000 (actual=00024000) [ECC]
ID=02           DJVPD 00091000..000d9000 (actual=00048000) [ECC]
ID=03            MVPD 000d9000..00169000 (actual=00090000) [ECC]
ID=04            CVPD 00169000..001b1000 (actual=00048000) [ECC]
ID=05        ATTR_TMP 01ca9000..01cb1000 (actual=00008000) 
ID=06       ATTR_PERM 01cb1000..01cb9000 (actual=00008000) [ECC]
ID=07           NVRAM 01de2000..01e72000 (actual=00090000) 
ID=08         FIRDATA 01e72000..01e75000 (actual=00003000) [ECC]
ID=09         BMC_INV 01e75000..01e7e000 (actual=00009000) 
ID=10         SECBOOT 01ea2000..01ec6000 (actual=00024000) [ECC]
ID=11             HBD 02008000..02068000 (actual=00060000) [ECC]
ID=12             HBI 02188000..02728000 (actual=005a0000) [ECC]
ID=13            SBEC 02728000..027b8000 (actual=00090000) [ECC]
ID=14             SBE 027b8000..02800000 (actual=00048000) [ECC]
ID=15            WINK 02800000..02920000 (actual=00120000) [ECC]
ID=16            HBRT 02920000..02c80000 (actual=00360000) [ECC]
ID=17         PAYLOAD 02c80000..02d80000 (actual=00100000) 
ID=18      BOOTKERNEL 02d80000..03c80000 (actual=00f00000) 
ID=19             OCC 03c80000..03da0000 (actual=00120000) [ECC]
ID=20            CAPP 03da0000..03dc4000 (actual=00024000) [ECC]
ID=21             HBB 03f67000..03ff7000 (actual=00090000) [ECC]
ID=22         VERSION 03ff7000..03ff8000 (actual=00001000) 
ID=23     BACKUP_PART 03ff8000..04000000 (actual=00000000) 
ID=24      OTHER_SIDE 00000000..00008000 (actual=00000000) 
About to erase chip !
Erasing... (may take a while !) Error -1 erasing chip

```

cyrilbur-ibm, 2016-11-04:
> Assigned #747 <https://github.com/openbmc/openbmc/issues/747> to
> @cyrilbur-ibm <https://github.com/cyrilbur-ibm>.

Please rerun with -g

I'll need to put more debugging in though I suspect.

shenki, 2016-11-07:
```
root@palmetto:~# pflash -g -E -f   
About to erase chip !
Erasing... (may take a while !) Error -1 erasing chip
```

cyrilbur-ibm, 2016-11-07:
Note to self need more debugging prints in that path. Definitely no MTD errors in dmesg? The -1 is suspicious and leads me to think the erase ioctl has failed and the problem is in the kernel. Another hint at this is the 4k erase granule, I thought that the kernel was favouring 64k erase blocks, is the kernel confused?

Another possibility is that pflash has tried to optimise to use a 64k erase but the kernel hasn't liked this, I would suggest that pflash is wrong to do this... erase granules has proven to be a bit of an issue as 4k erases don't work well when the kernel says it supports 64k erases. pflash was too tightly coupled with the flash driver it was initially written to work (and only work) with, I am addressing this.

I'll see if I can make a pflash to show this.

shenki, 2016-11-07:
```
open("/proc/mtd", O_RDONLY)             = 3
fstat64(3, {st_mode=S_IFREG|0444, st_size=0, ...}) = 0
read(3, "dev:    size   erasesize  name\nm"..., 1024) = 293
close(3)                                = 0
open("/dev/mtd7", O_RDWR)               = 3
fstat64(3, {st_mode=S_IFCHR|0600, st_rdev=makedev(90, 14), ...}) = 0
ioctl(3, MIXER_READ(1) or MEMGETINFO, {type=MTD_NORFLASH, flags=MTD_WRITEABLE|MTD_BIT_WRITEABLE, size=0x2000000, erasesize=0x10000, writesize=0x1, oobsize=0, padding=0}) = 0
ioctl(3, MIXER_READ(1) or MEMGETINFO, {type=MTD_NORFLASH, flags=MTD_WRITEABLE|MTD_BIT_WRITEABLE, size=0x2000000, erasesize=0x10000, writesize=0x1, oobsize=0, padding=0}) = 0
lstat64("/proc/self/fd/3", {st_mode=S_IFLNK|0700, st_size=64, ...}) = 0
readlink("/proc/self/fd/3", "/dev/mtd7", 65) = 9
fstat64(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 0), ...}) = 0
write(1, "About to erase chip !\n", 22About to erase chip !
) = 22
write(1, "Erasing... (may take a while !) ", 32Erasing... (may take a while !) ) = 32
write(2, "Error -1 erasing chip\n", 22Error -1 erasing chip
) = 22
```

cyrilbur-ibm, 2016-11-07:
Found the issue:

arch_flash_erase_chip() is (correctly...) bailing as it doesn't know how to erase a 'chip' opened via MTD, the procedure is simple but it just doesn't know how.

I'll have a patch by the end of the day.

cyrilbur-ibm, 2016-11-07:
This should solve (just patch 1/3): http://patchwork.ozlabs.org/patch/691765/

shenki, 2016-11-07:
Tested the fix on a plametto. Thanks @cyrilbur-ibm 

```
root@palmetto:/tmp# pflash -r pnor
Reading to "pnor" from 0x00000000..0x02000000 !
[==================================================] 100% ETA:0s     
root@palmetto:/tmp# pflash -e -p pnor 
About to erase 0x00000000..0x02000000 !
WARNING ! This will modify your HOST flash chip content !
Enter "yes" to confirm:yes
Erasing...
[==================================================] 100% ETA:0s     
About to program "pnor" at 0x00000000..0x02000000 !
Programming & Verifying...
[==================================================] 100% ETA:0s     
root@palmetto:/tmp# pflash -E -p pnor 
About to erase chip !
WARNING ! This will modify your HOST flash chip content !
Enter "yes" to confirm:yes
Erasing... (may take a while !) done !
About to program "pnor" at 0x00000000..0x02000000 !
Programming & Verifying...
[==================================================] 100% ETA:0s     
```

closed 2016-11-07 by williamspatrick