openbmc/openbmc #748: Romulus Power-On

mine260309, 2016-11-04:
Similar to #647, Romulus shall use a script to power on the system, with different steps.
The steps are as follow:
1. Power system up: 
  ```
gpioutil -p D1 -d out -v 1
  ```
2. FSI workaround toenable Nest mesh clock to clear out potential high VCS current draw on master processor
  ```
putcfam pu 2810 15 1 0 -ib       # Unfence PLL controls                            
putcfam pu 281A  1 1 1 -ib       # Assert Perv chiplet endpoint reset, just in case
putcfam pu 281A 31 1 1 -ib       # Enable Nest PLL      
  ```
 3. Tell CPLD to continue the power on sequence
  ```
gpioutil -p S7 -d out -v 0
  ```

williamli80, 2016-11-07:
@williamspatrick  @geissonator Maybe @mine260309 or @SS-IBM  can take this? Refers to #647

mine260309, 2016-12-21:
Witherspoon's power on scripts are merged (aac4adf831eebc74d97076e94125ebf5b6d5da48)
I will follow this commit to add Romulus' power on scripts.

Additionally, AVSBus and VRM related workarounds are needed for Romulus as well, so I will follow https://gerrit.openbmc-project.xyz/#/c/1567/ to add this workaround to Romulus.

mine260309, 2016-12-27:
Romulus power on sequence with AVSBus and VRM workaround:
1. avs_workaround
2. avsbus_disable
3. vrm_control
4. avs_enable
5. Power on
6. vcs_workaround
7. cpld_trigger (`gpioutil -p S7 -d out -v 0`)
8. start_host

mine260309, 2016-12-27:
Review:
https://gerrit.openbmc-project.xyz/#/c/1612
https://gerrit.openbmc-project.xyz/#/c/1613
https://gerrit.openbmc-project.xyz/#/c/1614

mine260309, 2016-12-28:
On Romulus, one problem is found that before power on (step 5 above), all operations on i2c-3,4,5 get timeout.
So the steps 1~4 won't work before power on, the sequence may need to be changed to:
1. Power on
2. avs_workaround
3. avsbus_disable
4. vrm_control
5. avs_enable
6. vcs_workaround
7. cpld_trigger (gpioutil -p S7 -d out -v 0)
8. start_host

But I'm not sure if this will work or not.

I will check with HW engineers to see if it's a bug or expected behavior.

mine260309, 2017-01-19:
The scripts of the workarounds are committed.
The left work:
1. Move "power on" step to the first step, because all the i2c operations require 3.3V power which is only available after "power on";
2. Update vcs_workaround and start_host scripts as pdbg is updated to specific the target (-a or -p0).

mine260309, 2017-02-13:
Extra changes are needed, that to pick the changes committed to Witherspoon but not to Romulus:
1. mbox daemon (and LPC-SPI related configs)
2. phosphor-hwmon-mrw
3. devtree-config

mine260309, 2017-02-15:
Except for the phosphor-hwmon-mrw and devtree-config changes (these should not block bring up), below are implemented and submitted to review:

* Move "power on" step to the first step and adjust vcs_workaround
https://gerrit.openbmc-project.xyz/#/c/2442
https://gerrit.openbmc-project.xyz/#/c/2446
* Power off script:
https://gerrit.openbmc-project.xyz/#/c/2445
* mbxox:
https://gerrit.openbmc-project.xyz/#/c/2443
The kernel device tree changes are reviewed in openbmc mailing list 

And an additional commits are made to update power on sequence with new obmc-power-chassis targets
https://gerrit.openbmc-project.xyz/#/c/2444/1

mine260309, 2017-02-27:
All the changes are merged.
Close the issue.

closed 2017-02-27 by mine260309

1997LP, 2021-09-28:
Where there are selling boards for romulus