openbmc/openbmc #749: obmc-console-server not starting

shenki, 2016-11-07:
```
root@palmetto:~# udevadm info -q all /dev/ttyS0 
P: /devices/platform/ahb/ahb:apb/1e787000.vuart/tty/ttyS0
N: ttyS0
E: DEVNAME=/dev/ttyS0
E: DEVPATH=/devices/platform/ahb/ahb:apb/1e787000.vuart/tty/ttyS0
E: MAJOR=4
E: MINOR=64
E: SUBSYSTEM=tty
E: TAGS=:systemd:
E: USEC_INITIALIZED=22674274
```

```
root@palmetto:~# systemctl status dev-ttyVUART0.device --no-pager
● dev-ttyVUART0.device
   Loaded: loaded
   Active: inactive (dead)

Nov 07 05:43:37 palmetto systemd[1]: dev-ttyVUART0.device: Job dev-ttyVUART0.device/start timed out.
Nov 07 05:43:37 palmetto systemd[1]: Timed out waiting for device dev-ttyVUART0.device.
Nov 07 05:43:37 palmetto systemd[1]: dev-ttyVUART0.device: Job dev-ttyVUART0.device/start failed with result 'timeout'.
```

This just sits there without returning:
```
root@palmetto:~# systemctl start dev-ttyVUART0.device
```

This is the reason udev does not trigger, as the lpc address is 0x417 when we expect 0x3f8:

```
root@palmetto:~# cat /sys/devices/platform/ahb/ahb\:apb/1e787000.vuart/lpc_address 
0x417
```

- `obmc-console-server` is launched when `/dev/ttyVUART0` is created.
- `/dev/ttyVUART0` is created when a device with `lpc_address` of `0x3f8` is created
- `lpc_address` is configured by obmc-console-server.
- GOTO 10;

A fix would be to match the udev rule on the address of the peripheral. This is what Cedric's original patch did:

```
SUBSYSTEM=="tty", ATTR{iomem_base}=="0x1E787000", SYMLINK+="tty-host"
```

shenki, 2016-11-07:
Original issue: https://github.com/openbmc/openbmc/issues/528#issuecomment-243356371

Potential dupes #718 #706

shenki, 2016-11-07:
https://gerrit.openbmc-project.xyz/#/c/1016/

geissonator, 2016-11-07:
@adamliyi - You've been our console issue guy recently, please take a look.

adamliyi, 2016-11-07:
OK. I think @shenki has fixed those tty timeout issues. I've add myself to https://gerrit.openbmc-project.xyz/#/c/1016/ reviewer.
Just a quick question: Why in kernel we change LPC address from *0x3f8* to *0x417*? And I do not understand the relationship between *ATTR{iomem_base}== "0x1E787000"* and address *0x417*? Could you point me to the kernel patch?

shenki, 2016-11-07:
We don't change that value in the kernel at all, and never have. It is read from the hardware at boot, and the documentation defines it as `x` (don't care) at reset. The value is supposed to be initialised by userspace and is currently done so by `obmc-console-server` in our OpenBMC.

`obmc-console-server` is started when it's character device (`/dev/tty*`) is created.

In the past we did not specify what the `/dev/` device should be called (eg in the device tree with an alias), the kernel is free to assign them how it pleases. When we upgraded the kernel from 4.4 to 4.7 the probing order changed, and the vuart moved from ttyS5 to ttyS0. In order to fix this a udev rule was written to symlink the vuart tty device to `/dev/ttyVUART0`. The udev rule needs to key off of something, so the lpc address attribute was chosen.

Now `obmc-console-server` was to start when `/dev/ttyVUART0` was created. It would then configure attributes `lpc_address`, `sio`, and open the chardev.

However, the creation of `ttyVUART0` depends on the value of `lpc_address`, which won't be set until `obmc-console-server` is started. The dependencies are around the wrong way.

`iomem_base` is the physical MMIO address that the Aspeed SoC has wired up for the uart. We should have always used something that is fixed in the hardware to key the udev rule from.

There's no kernel patch per se. You can see the source for the driver at https://github.com/openbmc/linux/blob/dev-4.7/drivers/tty/serial/aspeed-vuart.c.

This bug as been around ever since we switched to triggering obmc-console-server by the vuart symlink (1ba238171965c7d4773f5726b3a02c114b6f1831). If you were lucky enough to boot an old openbmc image which initialised the registers, they would be retained by the hardware until it lost power for future boots to use.

adamliyi, 2016-11-07:
@shenki . Thanks Joel for the detail comment- it clarified my doubt - why this timeout issue not always happen.

closed 2016-11-07 by williamspatrick