openbmc/openbmc #750: [ Barreleye v1.99.0-220-ge7b0fd5 ] System went for maintenance mode post reboot

gkeishin, 2016-11-07:
```  
Post update and some CI/CT test, a simple reboot crashed the system into maintenance mode.
Looks like something to do with file system corruption ? 
```

jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a0024: 0x7974 instead
jffs2: Further such events for this erase block will not be printed
jffs2: Cowardly refusing to erase blocks on filesystem with no valid JFFS2 nodes
jffs2: empty_blocks 63, bad_blocks 0, c->nr_blocks 64
mount: mounting /dev/mtdblock6 on run/initramfs/rw failed: Input/output error

Mounting read-write /dev/mtdblock6 filesystem failed.  Please fix and run
        mount /dev/mtdblock6 run/initramfs/rw -t jffs2 -o rw
to to continue, or do change nothing to run from RAM for this boot.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for system maintenance
(or type Control-D for normal startup):random: nonblocking pool is initialized

**Console output file attached  here for reference**
[Reboot Crashed.txt](https://github.com/openbmc/openbmc/files/575183/Reboot.Crashed.txt)

shenki, 2016-11-07:
```
rofs = mtd5 squashfs rwfs = mtd6 jffs2
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a0000: 0x7268 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a0004: 0x223d instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a0008: 0x6174 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a000c: 0x2f63 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a0010: 0x3036 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a0014: 0x3834 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a0018: 0x7661 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a001c: 0x6e6f instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a0020: 0x6f63 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x002a0024: 0x7974 instead
jffs2: Further such events for this erase block will not be printed
jffs2: Cowardly refusing to erase blocks on filesystem with no valid JFFS2 nodes
jffs2: empty_blocks 63, bad_blocks 0, c->nr_blocks 64
mount: mounting /dev/mtdblock6 on run/initramfs/rw failed: Input/output error

Mounting read-write /dev/mtdblock6 filesystem failed.  Please fix and run
        mount /dev/mtdblock6 run/initramfs/rw -t jffs2 -o rw
to to continue, or do change nothing to run from RAM for this boot.
```

I suggest you do as the text suggests: reflash your system from within the initramfs environment it has dropped you to.

The kernel version you are running is https://github.com/openbmc/linux/commit/c5682cbc3542add705e35152329436067bb1f94a which has the bad spi-nor patch reverted, so there should not be any persisting issue.

gkeishin, 2016-11-07:
The Serial console and network are gone for toss to do those.. Probably need to ask lab folks to recover..

gkeishin, 2016-11-10:
This is not seen.. Will re-open if further encountered

closed 2016-11-10 by gkeishin