openbmc/openbmc #751: Witherspoon: some I2C12 devices not working with new code

yadlapati, 2016-11-07:
Paul Robertson is seeing this issue:

Having problems with some devices on I2C12 after loading latest BMC image.  Being unable to talk to UCD chip, dev 0x64, causing problems with power on script.

$ ssh root@w1
root@w1's password:

root@witherspoon:~# i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- 0c -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

root@witherspoon:~# uname -r
4.7.10-c5682cbc3542add705e35152329436067bb1f94a

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-216"
VERSION_ID="v1.99.0-216-gcecba5f-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-216"
BUILD_ID="v1.99.0"

Following is the output (correct one) with older driver:
root@w3:~# i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- 0c -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- 32 -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- 51 -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- 64 -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@w3:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="0.1.0 (krogoth)"
VERSION_ID="v0.6-188-g2ad67f6-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (krogoth)"
BUILD_ID="v0.6"
root@w3:~#

adamliyi, 2016-11-07:
This may not be an bug. May be related with https://github.com/openbmc/openbmc/issues/653
@mine260309 , what do you think?

shenki, 2016-11-07:
This is normal system behaviour when a driver is loaded and bound to the address on the bus. As the kernel driver is in control of the device, you see UU at the address when using i2cdetect.

We need to provide Paul with documentation on how to use the pmbus sysfs files to perform the tests that he is doing. What are these tests? Which bits of information is he looking for?

paulgrobertson, 2016-11-07:
Here is the UCD script containing i2c commands that are failing.   How do we do this with UCD driver installed?

#!/bin/sh
# #############################################################
# script to disable UCD VCS rails. 
echo Disable UCD control of VCS
i2cset -y 11 0x64 0xF7 0x00 i
i2cset -y 11 0x64 0xF8 0x15 0x6E 0x80 0x0A 0x00 0x00 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 i
i2cset -y 11 0x64 0xF7 0x01 i
i2cset -y 11 0x64 0xF8 0x15 0x16 0x80 0x14 0x00 0x00 0x20 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 i
i2cset -y 11 0x64 0xD2 0x09 0x3F 0xFF 0x2F 0x0A 0x00 0x06 0x00 0x00 0x02 i
i2cset -y 11 0x64 0x00 0x0E i
i2cset -y 11 0x64 0x02 0x1A i
i2cset -y 11 0x64 0x00 0x0F i
i2cset -y 11 0x64 0x02 0x1A i

mine260309, 2016-11-07:
@paulgrobertson I know a way to do the test, but it could be a bit complicated: remove `ucd90160@64` part from `arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts`, rebuild the kernel, and then do the above test.
Maybe @shenki have better ways.

shenki, 2016-11-10:
I think you're correct that the driver does not provide the functionality we need.

While I work through the issue, you can unbind the driver from the device. This allows you to use your i2cset command again. I assume once the host has been brought up you could re-bind the driver.

To unbind:
```
cd /sys/bus/i2c/drivers/ucd9000/
echo 0-0064 > unbind
```

To re-bind once you're finished with the sequence:
```
cd /sys/bus/i2c/drivers/ucd9000/
echo 0-0064 > bind
```

I could not test this, so I am guessing at the `0-0064` name. You should see a directory inside `/sys/bus/i2c/drivers/ucd9000/` with the correct name; use that if it's not `0-0064`.

shenki, 2016-11-10:
I emailed @paulgrobertson with these details

paulgrobertson, 2016-11-11:
unbind/bind work around verified successfully.

geissonator, 2016-11-16:
Once we provide official power on support in the code, will this i2c command be required anymore (i.e. is there anything to do with this bug or do we just need to get official power on support in)?

paulgrobertson, 2016-11-16:
Yes.  VCS problem not fixed until dd2.0

shenki, 2017-09-25:
A solution has been in place for the past several months

closed 2017-09-25 by shenki