openbmc/openbmc #752: BMP280 needs to be enabled in witherspoon.cfg

spinler, 2016-11-07:
@mine260309  I think you handle these?

mine260309, 2016-11-07:
Sure, I will take this task.

mine260309, 2016-11-07:
Review posted: https://gerrit.openbmc-project.xyz/#/c/1026/

And I did some investigation of the device, it should be on `0x77@i2c-3`, and the device name should be `bmp180`.

spinler, 2016-11-09:
@mine260309 Why is it called bmp180 vs  bmp280?  The 280 is the one on the board.

mine260309, 2016-11-09:
@spinler I tried to read the chip id from the register, it shows "0x55" which means it's `bmp180`.
And from the witherspoon dts's comment, it indicatees it's BMP180 as well.
So I think the chip should be BMP180 instead of BMP280. While the linux config is `CONFIG_BMP280` for both devices.

spinler, 2016-11-09:
The BMP180 hit end of life, so witherspoon is going to ship with the 280.  It's possible these early systems may have some old hardware with the bmp180 on it, since the card was shared with firestone/preGA garrison).

I assume the device driver will do the probe and handle the device that is actually there so we can handle both anyway?

mine260309, 2016-11-10:
Yup, the driver handles both bmp180 and bmp280， and the kernel config is `CONFIG_BMP280` anyway.
@williamspatrick Since the review is closed and merged, could you help to close this issue？

shenki, 2016-11-10:
@mine260309 did you verify that the driver is enabled?

You need to set CONFIG_IIO=y :)

mine260309, 2016-11-10:
Right, that was missed! Added and review at https://gerrit.openbmc-project.xyz/#/c/1102/

closed 2016-11-18 by williamspatrick