openbmc/openbmc #753: Add rsync to the image

shenki, 2016-11-08:
Requested by our users

williamspatrick, 2016-11-08:
Why is rsync preferred when scp is already provided?  Can we get more background.

Doesn't appear that busybox supports rsync so that means we'll need to get "real" rsync.  On my x86 box it is .5 MB in size.  Hopefully we can disable a bunch of options and make it much smaller on ARM.

mikey, 2016-11-08:
rsync does incremental updates which is useful when on the other side of the world. There's a reason rsync was invented in Australia..

shenki, 2016-11-08:
```
root@palmetto:/usr/bin# ls -la rsync 
-rwxr-xr-x    1 root     root        404932 Nov  8 04:41 rsync
```

williamspatrick, 2016-11-08:
```
/tmp/rsync/rsync-3.1.2 $ ./configure --disable-largefile --disable-locale --disable-iconv --disable-acl-support --disable-xattr-support --without-included-popt --without-included-zlib
/tmp/rsync/rsync-3.1.2 $ ls -la rsync
-rwxr-xr-x 1 iawillia iawillia 366032 Nov  8 05:50 rsync
/tmp/rsync/rsync-3.1.2 $ ls -la rsync.gz
-rwxr-xr-x 1 iawillia iawillia 184148 Nov  8 05:50 rsync.gz
```

I was able to get it a decent bit smaller by changing the configure options to this.  We should probably also do similar on openbmc.  Any concerns with any of these options?  '--without-included-zlib' made a big dent in the size by itself.

shenki, 2016-11-09:
Yeah I think it's okay. I assume we already include the zlib and libpopt in the image? (Otherwise the reduced binary size might be offset by the larger libs)

williamspatrick, 2017-01-17:
@shenki started a commit here: https://gerrit.openbmc-project.xyz/#/c/1029/1

Need someone to take this over and continue the rsync size reduction work mentioned above.

chinaridinesh, 2017-03-14:
With these config flags --disable-largefile --disable-locale --disable-iconv --disable-acl-support --disable-xattr-support --without-included-popt --without-included-zlib  in  meta-phosphor/common/recipes-devtools/rsync/rsync_3.1.2.bbappend file, the size of rsync binary could reduced to 348716 bytes. Exploring if there are any more flags that can be used.

chinaridinesh, 2017-03-16:
Added required changes with following patch: https://gerrit.openbmc-project.xyz/#/c/2925/

williamspatrick, 2017-03-25:
Closed via 1ddd805657ea2336826395f11847a3fba36050a4

closed 2017-03-28 by bradbishop