openbmc/openbmc #755: [Barreleye "v1.99.0-226-g068b41c"] Failed to start Restore Last State.

gkeishin, 2016-11-08:
Keep seeing this error in case of code update. 

``` Oct 20 19:59:10 barreleye systemd[1]: Failed to start Restore Last State. ```

So doe this means the Restore policy is not persistent during code update..

Journal log attached here 
[journal_restore.txt](https://github.com/openbmc/openbmc/files/577548/journal_restore.txt)

gkeishin, 2016-11-08:
@ratagupt  anything on the restore policy here to look at..

williamli80, 2016-11-08:
Last system state is stored in  SYS_STATE_FILE = '/var/lib/obmc/last-system-state'. 
system_manager.py will set the property of "system_last_state".
discover_system_state.py will read system_last_state. ...
During Code update, is SYS_STATE_FILE readable?

ratagupt, 2016-11-08:
@gkeishin:- restore last state is not a binary so I am confused with the above error message... 
Seems that there is problem in starting of discover_system_state.py...
Do you have the system in the same state?I saw the system which you were using but seems you reverted it to some other build.

ratagupt, 2016-11-08:
meta-phosphor/common/recipes-phosphor/system/obmc-mgr-state/obmc-mgr-state.service...Here we are starting the discover_system_state.py..

gkeishin, 2016-12-01:
@ratagupt Any update on this ?