openbmc/openbmc #758: Handle OCC error conditions in hwmon driver

eddiejames, 2016-11-09:
[] create hwmon sysfs files for error conditions, such as power/temperature throttle
[] test error conditions with mocked up data

amboar, 2016-11-09:
@eddiejames can you please provide more of a description for the first item? What is the behaviour of these sysfs files? When will they be present? Always? Only in the face of errors? Is sysfs the right interface for exposing this information? What about logging the problem (to the journal) instead?

eddiejames, 2016-11-10:
Sure @amboar. For these error conditions, the OCC spec says:
the BMC  should assert the “Processor Frequency Limited due to over temperature” sensor for this processor when this bit goes from ‘0’ to ‘1’ between two polls and de-assert when this bit goes from ‘1’ to ‘0’ between two polls.

All the error conditions are described in similar fashion. So that sounds to me that it would be best done with a sysfs file that is present only when that bit is asserted. The file could read out the whole status register, perhaps. We could ask Martha if logging would be a better approach I guess.

Some of the other conditions might be better suited for the log, from the spec, for example if the OCC is in the "safe" state.

amboar, 2016-11-14:
@eddiejames I'm not keen on the idea of sysfs files appearing/disappearing: Could we instead always present the files with the content of the poll result, e.g. call the attribute 'overtemp' and have it read 0 if false, 1 if true.

What are the consequences of asserting the sensor? Is the sysfs attr value the only thing that changes? Do we need to change power management policy on the BMC? I guess I'm trying to understand whether we'll have other processes polling these files as well. If so maybe we should do something more event-based.

geissonator, 2017-03-27:
Somewhat related to #1234

mdmillerii, 2017-03-29:
Userspace can `poll` for `sysfs` attribute content changes, if the driver makes an api call to `sysfs_notify` to inform `sysfs` the content has changed.

I agree with @amboar that exposing the value in a file that is present is better than creating a file on error.  In fact I'd expect it could just be another attribute for hwmon.

eddiejames, 2017-09-22:
this is done.

closed 2017-09-22 by eddiejames