openbmc/openbmc #759: [Barreleye] Fail to boot host because of pflash has stuck

johnhcwang, 2016-11-10:
Sometimes we got the pflash error message **LIBFLASH: WREN has WIP status set !** and **pflash occupied high CPU resource** during AC cycle test.
In this case, the state didn't be change to INVENTORY_UPLOADED, so BMC didn't start FSI command to boot host.

Hi @anoo1 , have you seen this situation? Is it possible if we merge some pflash updates since Sep 12 to fix this issue?

````
Nov 02 06:16:00 barreleye system_manager.py[948]: Starting flasher: bios,,/org/openbmc/control/flash/bios,
Nov 02 06:16:00 barreleye system_manager.py[948]: flasher: , BMC = 0, address = 0x0
Nov 02 06:16:00 barreleye system_manager.py[948]: Setting up BIOS flash
Nov 02 06:16:00 barreleye system_manager.py[948]: LIBFLASH: WREN has WIP status set !
Nov 02 06:16:00 barreleye system_manager.py[948]: Flash tuned
Nov 02 06:16:26 barreleye systemd[1]: systemd-hwdb-update.service: Cannot add dependency job, ignoring: Unit systemd-hwdb-update.service is masked.
````
````
 PID  PPID   USER      STAT     VSZ  %VSZ  %CPU   COMMAND
1081  1079   root      R      69768   27%   87%   pflash -f -e -p /tmp/tmpS4p_td  -P BMC_INV
````
[journal log.txt](https://github.com/openbmc/openbmc/files/582362/journal.log.txt)

williamspatrick, 2016-11-10:
@johnhcwang - This is a question with respect to the stable branch?  I see the following in your journal output, so I assume so.

```
Sep 11 12:06:13 barreleye kernel: Linux version 4.4.18-v10.60.2560 (jason@ken-PowerEdge-R210-II) (gcc version 4.9.3 (GCC) ) #1 Wed Nov 2 10:32:52 CST 2016
```

johnhcwang, 2016-11-10:
@williamspatrick - Yes, the issue we found on the v1.0-stable.

johnhcwang, 2016-11-14:
@williamspatrick @anoo1 - Is it probably related to #713 ?

shenki, 2016-11-15:
@johnhcwang This is not related to those issues; the code base has changed a lot between master and the 1.0-stable branch.

Can you please tell me the model of the SPI NOR chip used in this system?

The output of `pflash -i` will tell you this.

willenlee, 2016-11-15:
Hi Shenki

The output data is shown below.
```
root@barreleye:~# pflash -i
Flash info:
-----------
Name          = Micron N25Qx512Ax
Total size    = 64MB
Erase granule = 4KB

TOC@0x00000000 Partitions:
-----------
ID=00            part 00000000..00001000 (actual=00001000)
ID=01            HBEL 00008000..0002c000 (actual=00024000) [ECC]
ID=02           GUARD 0002c000..00031000 (actual=00005000) [ECC]
ID=03             HBD 00031000..00091000 (actual=00060000) [ECC]
ID=04           DJVPD 00091000..000d9000 (actual=00048000) [ECC]
ID=05            MVPD 000d9000..00169000 (actual=00090000) [ECC]
ID=06            CVPD 00169000..001b1000 (actual=00048000) [ECC]
ID=07             HBI 001b1000..00751000 (actual=005a0000) [ECC]
ID=08            SBEC 00751000..007e1000 (actual=00090000) [ECC]
ID=09             SBE 007e1000..00829000 (actual=00048000) [ECC]
ID=10            WINK 00829000..00949000 (actual=00120000) [ECC]
ID=11            HBRT 00949000..00ca9000 (actual=00360000) [ECC]
ID=12         PAYLOAD 00ca9000..00da9000 (actual=00100000)
ID=13      BOOTKERNEL 00da9000..01ca9000 (actual=00f00000)
ID=14        ATTR_TMP 01ca9000..01cb1000 (actual=00008000)
ID=15       ATTR_PERM 01cb1000..01cb9000 (actual=00008000) [ECC]
ID=16             OCC 01cb9000..01dd9000 (actual=00120000) [ECC]
ID=17            TEST 01dd9000..01de2000 (actual=00009000) [ECC]
ID=18           NVRAM 01de2000..01e72000 (actual=00090000) [ECC]
ID=19         FIRDATA 01e72000..01e75000 (actual=00003000) [ECC]
ID=20         BMC_INV 01e75000..01e7e000 (actual=00000000)
ID=21            CAPP 01e7e000..01ea2000 (actual=00024000) [ECC]
ID=22         SECBOOT 01ea2000..01ec6000 (actual=00024000) [ECC]
ID=23             HBB 01f60000..01ff0000 (actual=00090000) [ECC]
ID=24         VERSION 01ff7000..01ff8000 (actual=00001000)
ID=25     BACKUP_PART 01ff8000..02000000 (actual=00000000)
ID=26      OTHER_SIDE 02000000..02008000 (actual=00000000)

TOC@0x02000000 Partitions:
-----------
ID=00            part 02000000..02001000 (actual=00001000)
ID=01            HBEL 00008000..0002c000 (actual=00024000) [ECC]
ID=02           DJVPD 00091000..000d9000 (actual=00048000) [ECC]
ID=03            MVPD 000d9000..00169000 (actual=00090000) [ECC]
ID=04            CVPD 00169000..001b1000 (actual=00048000) [ECC]
ID=05        ATTR_TMP 01ca9000..01cb1000 (actual=00008000)
ID=06       ATTR_PERM 01cb1000..01cb9000 (actual=00008000) [ECC]
ID=07           NVRAM 01de2000..01e72000 (actual=00090000) [ECC]
ID=08         FIRDATA 01e72000..01e75000 (actual=00003000) [ECC]
ID=09         BMC_INV 01e75000..01e7e000 (actual=00009000)
ID=10         SECBOOT 01ea2000..01ec6000 (actual=00024000) [ECC]
ID=11             HBD 02008000..02068000 (actual=00060000) [ECC]
ID=12             HBI 02188000..02728000 (actual=005a0000) [ECC]
ID=13            SBEC 02728000..027b8000 (actual=00090000) [ECC]
ID=14             SBE 027b8000..02800000 (actual=00048000) [ECC]
ID=15            WINK 02800000..02920000 (actual=00120000) [ECC]
ID=16            HBRT 02920000..02c80000 (actual=00360000) [ECC]
ID=17         PAYLOAD 02c80000..02d80000 (actual=00100000)
ID=18      BOOTKERNEL 02d80000..03c80000 (actual=00f00000)
ID=19             OCC 03c80000..03da0000 (actual=00120000) [ECC]
ID=20            CAPP 03da0000..03dc4000 (actual=00024000) [ECC]
ID=21             HBB 03f67000..03ff7000 (actual=00090000) [ECC]
ID=22         VERSION 03ff7000..03ff8000 (actual=00001000)
ID=23     BACKUP_PART 03ff8000..04000000 (actual=00000000)
ID=24      OTHER_SIDE 00000000..00008000 (actual=00000000)
```

anoo1, 2016-11-15:
@shenki Do you think it'd help to update the stable branch to the upstream pflash like master, or what would you recommend for this issue?

shenki, 2016-11-15:
@anoo1 It my help. I don't know what the issue is though, so it may not.

You would need to backport the patches that refactored flasher to use libflash, as well as the pflash/libflash bitbake configuration.

williamspatrick, 2016-11-15:
Based on the usage pointed out here, we are directly using 'pflash' in this error path.  On boot we update the "BMC_INV" partition with a device tree using pflash.  Since this is the [only] heavily used pflash call done it might be straight forward to move pflash and leave flasher as is to [hopefully] solve this?

gtmills, 2018-05-23:
@johnhcwang @shenki Still a problem? Can we close?

closed 2018-05-24 by shenki