openbmc/openbmc #760: [Barreleye VERSION_ID="v1.99.0-237-gb02da7b] Failed to start Phosphor IPMI BT DBus Bridge.

gkeishin, 2016-11-10:
[  OK  ] Started Clear one time boot overrides.
```
[FAILED] Failed to start Phosphor IPMI BT DBus Bridge.
See 'systemctl status org.openbmc.HostIpmi.service' for details.
[  OK  ] Stopped Phosphor IPMI BT DBus Bridge.
         Starting Phosphor IPMI BT DBus Bridge...
         Starting Phosphor Inband IPMI...
[FAILED] Failed to start Phosphor IPMI BT DBus Bridge.
See 'systemctl status org.openbmc.HostIpmi.service' for details.
[  OK  ] Stopped Phosphor IPMI BT DBus Bridge.
         Starting Phosphor IPMI BT DBus Bridge...
[FAILED] Failed to start Phosphor IPMI BT DBus Bridge.
See 'systemctl status org.openbmc.HostIpmi.service' for details.
```
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 barreleye ttyS4

barreleye login: ftgmac100 1e660000.ethernet eth0: NCSI interface up
random: nonblocking pool is initialized

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 barreleye ttyS4

barreleye login: root
Password:
root@barreleye:~#

● org.openbmc.HostIpmi.service - Phosphor IPMI BT DBus Bridge
   Loaded: loaded (/lib/systemd/system/org.openbmc.HostIpmi.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: exit-code) since Thu 2016-10-20 19:57:53 UTC; 5min ago
  Process: 717 ExecStart=/usr/sbin/btbridged [[0;1;31m(code=exited, status=203/EXEC)[[0m
 Main PID: 717 (code=exited, status=203/EXEC)

Oct 20 19:57:53 barreleye systemd[1]: [[0;1;39morg.openbmc.HostIpmi.service: Main process exited, code=exited, status=203/EXEC[[0m
Oct 20 19:57:53 barreleye systemd[1]: [[0;1;31mFailed to start Phosphor IPMI BT DBus Bridge.[[0m
Oct 20 19:57:53 barreleye systemd[1]: [[0;1;39morg.openbmc.HostIpmi.service: Unit entered failed state.[[0m
Oct 20 19:57:53 barreleye systemd[1]: [[0;1;39morg.openbmc.HostIpmi.service: Failed with result 'exit-code'.[[0m
Oct 20 19:57:53 barreleye systemd[1]: org.openbmc.HostIpmi.service: Service hold-off time over, scheduling restart.
Oct 20 19:57:53 barreleye systemd[1]: Stopped Phosphor IPMI BT DBus Bridge.
Oct 20 19:57:53 barreleye systemd[1]: [[0;1;39morg.openbmc.HostIpmi.service: Start request repeated too quickly.[[0m
Oct 20 19:57:53 barreleye systemd[1]: [[0;1;31mFailed to start Phosphor IPMI BT DBus Bridge.[[0m

root@barreleye:~# 
**Journal log attached**

[barreleye_v1.99.0-237-gb02da7b_journal.txt](https://github.com/openbmc/openbmc/files/582543/barreleye_v1.99.0-237-gb02da7b_journal.txt)

shenki, 2016-11-10:
btbrdiged is not i nthe filesystem on my machine:

```
root@palmetto:~# ls -la /usr/bin/btbridged
ls: /usr/bin/btbridged: No such file or directory
root@palmetto:~# find / -name btbridged
root@palmetto:~# echo $?
0
```

What happened to it?

gkeishin, 2016-11-10:
On barreleye also i dont see that btbridged file anywhere...

shenki, 2016-11-10:
Fixed by reverting 7729fe6efcf72a3e3057f13c27c922bfac570196.

https://gerrit.openbmc-project.xyz/#/c/1093/

williamspatrick, 2016-11-10:
Resolved by cc1502f3d6c544116ed570b2a02da2cdcdd550c1

closed 2016-11-10 by williamspatrick