openbmc/openbmc #762: Witherspoon:  Reading 0's from PNOR  (LPC mapping support)

yadlapati, 2016-11-10:
We are doing Witherspoon bringup and seeing all 0's when tried to read from PNOR. Joel Stanley looked at the problem. There are two issues here:
1. We are using old driver. 
2. LPC mapping support  is needed for Witherspoon.

shenki, 2016-11-10:
When using the new driver, you can map the first 64MB of the PNOR into the LPC address space with the following:

```
devmem 0x1e789088 0x30000c00
devmem 0x1e78908c 0xffc0003f
devmem 0x1e789080 0x00000500
```

The PNOR will then appear at `0x60300fc000000` once the host has enabled the LPC.

I will implement this configuration in OpenBMC in the next day or two. This workaround is so you are not blocked on that work.

shenki, 2016-11-11:
Updated sequence from Ben:

```
devmem 0x1e789088 32 0x30000C00
devmem 0x1e78908c 32 0xFC0003FF
devmem 0x1e630034 32 0x70680000
devmem 0x1e630030 32 0x68600000
devmem 0x1e789080 32 0x00000500
```

@amboar Looks like we need a driver for the AHB controller too.

shenki, 2016-11-29:
As of https://github.com/openbmc/linux/commit/8139ec738741c7783fd5bcb4ce81453d87f47b17 we configure this in the kernel board file.

This is a temporary solution while we sort out the details of a proper driver.

This was included in 0d888ed572c5fa45c2ea3d24fe57cf385b83cc34.

yadlapati, 2017-01-13:
We are still seeing this issue on Witherspoon system (w1.aus.stglabs.ibm.com) with new driver.

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-383"
VERSION_ID="v1.99.0-383-g4a2201f-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-383"
BUILD_ID="v1.99.0"
root@witherspoon:~# uname -a
Linux witherspoon 4.7.10-120956cab11d86fa801d53811ed2f7f844216a2f #1 Tue Jan 10 22:38:52 UTC 2017 armv6l GNU/Linux

williamspatrick, 2017-01-18:
#890 is the same issue.

closed 2017-01-18 by williamspatrick