openbmc/openbmc #766: hostcheckstop refactor

geissonator, 2016-11-10:


mine260309, 2016-12-15:
The inteface `org.openbmc.control.Checkstop` seems to have no methods nor properties.
Is it meaningful to define an empty interface in `phosphor-dbus-interfaces`?

geissonator, 2016-12-15:
Discussed a bit in design meeting.  The thought is to not have a sensor or sdbus property for this.  We'll simply have a process that monitors the checkstop gpio, and issues a systemd transition request to reboot system when it hits.  Patrick proposed we rename this to something more generic like - "Host Termination Signal".  We may also need to create an elog when this happens.

spinler, 2016-12-15:
For checkstop debug we'll need a way to disable this.  Is it as easy as stopping the service?

geissonator, 2016-12-15:
Good point, yes, that should be a requirement here.

geissonator, 2016-12-15:
And also, probably shouldn't use systemd directly, should use the new Host interface to restart things (it calls systemd).

geissonator, 2017-02-06:
This will use the new quiesce target functions being created in #1033

vishwabmc, 2017-03-02:
Eddie is currently doing some GPIO support and I feel we can park this story for future sprints considering that its needed for phase-5 and gpio stuff is being updated.

@geissonator can we move this out ?

vishwabmc, 2017-03-07:
There was a design discussion on this and here is the key takeaway.

* 4.9 Kernel is coming with a IOCTL for GPIO which gives the FD for a passed in GPIO line.
* We can still go ahead and do the work now without waiting for 4.9
  - idea is to have a stub that will do a open on some existing filesystem entity ( like a fifo, pipe, file etc ).
  - and let that give the FD and when 4.9 is in, this would be augmented to using that GPIO IOCTL
* The application will be passed the gpio chip number and the gpio number so that the app does not do any
   interpretation.
* then the usual sd_event mechanism can be used to wait on the gpio assertions

* The application does not need to worry about setting the GPIO direction as its a IN pin.

vishwabmc, 2017-03-09:
- [ ]

geissonator, 2017-03-10:
Lets touch base in Monday call on what will happen when we hit the gpio.

sivassrr, 2017-03-13:
https://gerrit.openbmc-project.xyz/#/c/2938/ - Do we need to handle anything else in our test. 
(Hope this changes cover for our test, let know if we need to add any).  We can add new issue.

vishwabmc, 2017-03-24:
in yesterday's meeting, one of the idea was to use /dev/input/event* to monitor the GPIO line.

I looked at Palmetto / Barreleye / Witherspoon system and did not see /dev/input.

Looks like we don't have input device support in the Kernel ?

@bradbishop ,  looks like its wise to wait till we move to 4.10 and use the IOCTL ?.

@geissonator I see its marked for `Phase-5` and I believe this needs to be taken up when we are ready with the support ?

@shenki you have any thoughts ?

vishwabmc, 2017-03-24:
@rfrandse I see this issue has history of its ready for 17.02 etc and I feel its not appropriate since we don't even seem to have the needed support ?

geissonator, 2017-03-24:
This is needed for phase 4 so we can put the error creation code in (so external users are notified when we hit a checkstop).  #1266 is the story for that.

vishwabmc, 2017-03-24:
Needs https://github.com/openbmc/openbmc/issues/1387

shenki, 2017-03-26:
The suggestion relating to input devices was to use the GPIO keys driver:

https://www.kernel.org/doc/Documentation/devicetree/bindings/input/gpio-keys.txt

In addition entries to the device tree, you would need to enable `CONFIG_KEYBOARD_GPIO` in the kernel.

vishwabmc, 2017-03-28:
I talked to @bradbishop yesterday and he was saying that relaying the GPIO values onto /dev/input/ needs some work there. We would be discussing on that in today's design meeting.

@shenki if you have any thought, please let me know.

closed 2017-04-27 by williamspatrick