openbmc/openbmc #767: hostwatchdog refactor

geissonator, 2016-11-10:


adamliyi, 2016-12-01:
- [ ] Spend 2-3 days study led-manager to get to know with sdbusplus
- [ ] Define an interface (yaml) for host watchdog module - prefer properties than method
- [ ] Implement the new host watchdog dbus object with the new interface (may utilize existing code)

adamliyi, 2016-12-02:
Add the boot watchdog interface in patch: https://gerrit.openbmc-project.xyz/#/c/1370/.

adamliyi, 2016-12-15:
Patch committed here: https://gerrit.openbmc-project.xyz/#/c/1513/

geissonator, 2017-01-05:
This was not fully resolved, implementation still needs to be done in https://gerrit.openbmc-project.xyz/#/c/1513/

geissonator, 2017-05-15:
Time is out on this one, need it for mfg stage 3.  Vishwa to help finish it up this sprint.

vishwabmc, 2017-05-18:
Revised task-list

- [ ] Update `Watchdog.interface.yaml` since there is no signal being sent on timeout.
- [ ] Write command line parser. There are 3 arguments to be passed now.
- [ ] Write Timer class providing timer support using sd_event
- [ ] Write Watchdog class to use Timer class and implement the interfaces.
- [ ] Create a new systemd.target that will be called by this application
- [ ] Make changes to openpower-debug-collector to add new error definitions
- [ ] Associate openpower-debug-collector with the newly created systemd.target
- [ ] Create recipe file for this with using ENV variables
- [ ] Add Gtest cases

rfrandse, 2017-06-01:
https://gerrit.openbmc-project.xyz/4530 Commit error and exit the application on error scenarios
Fixes #767 hostwatchdog refactor

closed 2017-06-08 by williamspatrick