openbmc/openbmc #770: pciedetect refactor

geissonator, 2016-11-10:


williamli80, 2016-12-02:
- [ ] Find out the yaml ( where is it? ) and generate the sdbusplus binding.
- [ ] Implement the pcie_slot_present with new binding ( such as update_fru( ) use new APIs/dbus )
- [ ] Update phosphor-host-ipmid repo

right?

geissonator, 2016-12-12:
This is focused on updating the new inventory objects.

williamli80, 2016-12-14:
-[ ] Creating new inventory objects to inventory manager.
-[ ] Implementing pciedetect 
-[ ] calling new sdbusplus interface to update them.
Use above created inventory objects, read gpio signal, update gpio inventory objects.

geissonator, 2016-12-15:
@spinler to check if we even wired up Witherspoon to support this function.

geissonator, 2016-12-15:
This will require the sdbusplus client bindings to talk with inventory manager.

geissonator, 2016-12-15:
- interrupt driven
- Make generic inventory object that is created when a GPIO is asserted (indicates presence)
- MRW information drives this program (MRW -> service files that launch inventory object)
Talk with Matt S. and Brad for details.

williamli80, 2017-01-15:
As we discussed, because we miss the feature of GPIO in new format, currently we do following steps:
- [ ] write a program waiting for a file descriptor with sd_event.
- [ ] When data of the file descriptor changed, we make a dbus call to update the inventory item.

(Update the common file descriptor when we get new GPIO support.)
@geissonator @bradbishop @williamspatrick

closed 2019-10-27 by stale[bot]