openbmc/openbmc #781: Map IPMI sensors to appropriate inventory objects

geissonator, 2016-11-10:


geissonator, 2017-01-05:
This needs to at least support the hostboot sensor information (i.e. core functional).
This will just be functional state sensors (Andrew create new story for rest of work)

geissonator, 2017-01-23:
This has a dependency on some work from Brad to allow state to be tracked in inventory.  Only putting this in actual (vs. committed) for now due to this.

spinler, 2017-02-06:
At a high level, this will need to map the 'present' and 'disabled' sensor assertions for the DIMM and core functional sensors to inventory operations:
* when a core functional sensor has the Present bit asserted, offset 7 = 0x0080, the core needs to be created in the inventory
* when a core functional sensor has the Disabled bit asserted, offset 8 = 0x0100, the functional property on the inventory object should be set to false.  I'm not sure if this interface/property already exists. 
* the present property should be True if the disabled bit isn't set (maybe default to true?)
* DIMM inventory objects are created by the FRU data hostboot sends down, so this code doesn't need to create DIMM inventory objects ever.  It just needs to set the Functional Property to false if the Disabled bit is set in the sensor (offset 4 = 0x0010)

vishwabmc, 2017-02-08:
Please reassign this as per discussion in scrum

ratagupt, 2017-02-21:
- [ ]  Fetching the data from MRW.

- [ ] YAML design.

- [ ] IPMI sensor handler changes.

- [ ] Recipe changes

ratagupt, 2017-02-23:
- [ ] Generate the cpp from YAML.

ratagupt, 2017-02-23:
code is up for review for the checked task.

closed 2017-03-16 by williamspatrick