openbmc/openbmc #783: Host state management and control

geissonator, 2016-11-10:


geissonator, 2016-11-17:
[ ] - Determine requirements, send out email to distribution list
[ ] - Get new repo created, generate recipe and makefiles
[ ] - Define dbus interfaces (methods and properties)
[ ] - Implement properties
[ ] - Implement methods
[ ] - Write test cases to verify new properties and methods

Note that #784 will be where we officially move everything over to the new state management interfaces

geissonator, 2016-12-01:
As I've dug more and more into implementation, it's become clear we need a long term vision for our system/chassis/host/bmc state management and control.  I've tried to step back and look at the use cases for these areas so we can setup a long term direction.

**System/Host Control Use Cases**

### Primary Use Cases ###
- Determine state of system (bmc, chassis power, host)
-- Must be valid at any arbitrary instance in time
- Boot the system (host)
- Reboot the system (host)
-- Hard (cut power, boot back up) and Soft (notify host of shutdown)
- Just power on system, but not boot the host (debug mode)
- Reboot the BMC

### Secondary Use Cases ###
- Enable and handle host watchdog
- Verify pgood
- Handle signals for power buttons (on, off, emergency power off)
- Handle host checkstop
- Allow user to set a policy (auto power on/off or stay same) and enforce it on BMC reboots

### Implementation ###
- Maintain system (host) state over BMC reboots
- Run FSI bit bang 
- Do pcie and usb resets on power on/off

```
Current SYSTEM_STATES

SYSTEM_STATES = [
	'BASE_APPS',
	'BMC_STARTING',
	'BMC_STARTING2',
	'BMC_READY',
	'HOST_POWERING_ON',
	'HOST_POWERED_ON',
	'INVENTORY_UPLOADED',
	'HOST_BOOTING',
	'HOST_BOOTED',
	‘HOST_POWERED_OFF']
```

yuennancy, 2016-12-01:
What is the difference between BMC_STARTING vs BMC_STARTING2?
What does INVENTORY_UPLOADED refer to? Firmware updates?

What's the difference between HOST_POWERING/ED_ON vs HOST_BOOTING?ED? 
BOOT refers to executing code (bios, kernel) and POWER refers to everything before?

geissonator, 2016-12-01:
Hah, yeah good questions @yuennancy .  My hope with this refactoring is to simplify things a bit if possible.  If you dig in, BMC_STARTING and STARTING2 represent different objects being started  as we get to BMC_READY.  You can see the details in here on what objects cause which state - https://github.com/openbmc/skeleton/blob/master/configs/Barreleye.py#L18

I believe INVENTORY_UPLOADED is once Hostboot has sent down all their known inventory to the BMC.  

HOST_POWERING?ED represents the turning on of chassis power and pgood being asserted (indicating power really is on).  I think HOST_BOOTING?ED is hostboot in the process of booting and it being done booting (i.e. petitboot ready).

amboar, 2016-12-02:
BMC_STARTING2 seems like an implementation detail. If it's not, maybe some more effort should be (have been?) put into the name.

yuennancy, 2016-12-02:
Thanks for clarifying these.

closed 2017-01-18 by geissonator