openbmc/openbmc #791: Create udev-rule generating recipe

bradbishop, 2016-11-11:
Write a recipe that generates udev rules/systemd-device units which launch instances of phosphor-hwmon for hwmon entries.

spinler, 2016-11-16:
[ ] - Write a script that generates udev rules/systemd-device units from MRW which launch instances of phosphor-hwmon for hwmon entries.
[ ] - Create a native recipe for script.
[ ] - Create a target recipe that runs the script.
[ ] - Create hwmon udev-rule generating recipes for existing non-mrw systems.
[ ] - Enhance hwmon udev rules to include configuration (thresholds)
[ ] - Add hwmon<->udev virtual and set PREFERRED_PROVIDER for each system

spinler, 2016-11-17:
Writing down some thoughts...
There is only 1 parameter that can be passed to a systemd unit (the template parameter) from udev, so how to get multiple pieces of information - the device path, the ~2 threshold values, and possibly a common name to the unit...

Ideas:
1. Send in the name of EnvironmentFile file that has been previously generated from MRW data (named by I2C bus/addr) that holds the thresholds.   Then the .service has:  EnvironmentFile=%I  But then, how to get the device path? (Have udev rule add it to the env file first?)
2.  Have udev rule store env file name in a variable, then pass the device path to the service, and have ```ExecStart = phosphor-hwmon %I `udevadm info %I | grep ENVFILE | sed s/E: ENVFILE=//'```
But, then can't use "EnvironmentFile= " line in the unit file so phosphor-hwmon has to read the file.
3.  Like 2, but instead start a wrapper script that calls udevadm info <devpath>, parses the output to get the env file name, pulls out its values, and sends them all into phosphor-hwmon on the command line

amboar, 2016-11-22:
Might be a long shot as the restrictions are quite tight, but can we leverage systemd unit generators?

spinler, 2016-11-22:
Prosposed udev rules at https://gerrit.openbmc-project.xyz/#/c/1243/

spinler, 2016-11-22:
@amboar Could you give an example of how that would work?

amboar, 2016-11-23:
@spinler I didn't have anything concrete in mind, just thought it might be worth asking the question.

geissonator, 2017-01-23:
https://gerrit.openbmc-project.xyz/#/c/1835/

closed 2017-01-24 by geissonator