openbmc/openbmc #794: [ Barreleye v1.99.0-242-gfddb958 ] Poweron didn't succeed to Host booted state

gkeishin, 2016-11-11:
root@barreleye:~# obmcutil state
 = HOST_POWERED_ON

Suspecting something to do with inventory ... 
```
Oct 20 20:00:29 barreleye systemd[1]: Started Inventory Upload.
Oct 20 20:00:30 barreleye inventory_upload.py[851]: ERROR: Input tree has errors, aborting (use -f to force output)
````
- [x] Journal Log  - 
[barreleye_v1.99.0-242-gfddb958.txt](https://github.com/openbmc/openbmc/files/585783/barreleye_v1.99.0-242-gfddb958.txt)

geissonator, 2016-11-11:
This looks new:
```
Oct 20 20:00:18 barreleye systemd[1]: Started Read system/chassis/io_board EEPROM.
Oct 20 20:00:18 barreleye systemd[1]: Starting Sync system/chassis/io_board...
Oct 20 20:00:24 barreleye sync_inventory_items.py[883]: Rebooting BMC to set uuid
Oct 20 20:00:24 barreleye systemd[1]: Started Sync system/chassis/motherboard.
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]: Traceback (most recent call last):
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:   File "/usr/sbin/sync_inventory_items.py", line 143, in <module>
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:     sync_mac(net_obj, inv_value, sys_mac)
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:   File "/usr/sbin/sync_inventory_items.py", line 81, in sync_mac
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:     dbus_method("eth0", mac_str)
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:     **keywords)
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:     message, timeout)
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]: dbus.exceptions.DBusException: org.freedesktop.DBus.Python.IOError: Traceback (most recent call last):
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:   File "/usr/lib/python2.7/site-packages/dbus/service.py", line 707, in _message_cb
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:     retval = candidate_method(self, *args, **keywords)
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:   File "/usr/sbin/netman.py", line 392, in SetHwAddress
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:     self.validatemac(mac)
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:   File "/usr/sbin/netman.py", line 166, in validatemac
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:     int_eep_mac = int(get_mac_from_eeprom(), 16)
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:   File "/usr/sbin/netman.py", line 112, in get_mac_from_eeprom
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:     bus_name, path, intf, prop = read_mac_provider_info()
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:   File "/usr/sbin/netman.py", line 83, in read_mac_provider_info
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]:     raise IOError("Could not find %s" % conf_file)
Oct 20 20:00:25 barreleye sync_inventory_items.py[890]: IOError: Could not find /etc/network-manager.conf
Oct 20 20:00:25 barreleye systemd[1]: obmc-sync-inventory-item@system-chassis-io_board.service: Main process exited, code=exited, status=1/FAILURE
Oct 20 20:00:25 barreleye systemd[1]: Failed to start Sync system/chassis/io_board.
```

gkeishin, 2016-11-11:
This is seen in latest build as well v1.99.0-244-g11381c4

amboar, 2016-11-14:
Aren't we using systemd-networkd now?

williamli80, 2016-11-14:
@amboar I think so. systemd-networkd is running.
IN BarE-01:
william@ubuntu:~$ ssh root@9.3.29.119
root@9.3.29.119's password:
root@barreleye:~# ps|grep -i systemd-networkd
 1191 systemd-   0:00 /lib/systemd/systemd-networkd
 1882 root       0:00 grep -i systemd-networkd
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-245"
VERSION_ID="v1.99.0-245-g49dea7a-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-245"
BUILD_ID="v1.99.0"

williamli80, 2016-11-14:
fddb958f0a7ce2d3296b934fa0b438de3ab011ad involved the error. This commit just bumped version of  phosphor-settings repository. So I think maybe phosphor-settings is the source of the error. 
I'm looking at 62b55f35fde5e057bb7b0294f2e2d8ad66170e2d and 0fdfa72ee81bf52ba65bfc401d85d0e491abbf40.

williamli80, 2016-11-14:
#806

williamspatrick, 2016-11-15:
Does that mean this is a dupe of 806?

gkeishin, 2016-11-16:
Still seen on v1.99.0-251-gb78c6de latest build.. #806 is now fixed and verified .. but this poweron on BARE01 is still not good.. will try on another BarPvt to confirm if this still lingers on

gkeishin, 2016-11-17:
Looks good on v1.99.0-257-g85b85b3  ..did someone fix anything or it just went off and worked ?

gkeishin, 2016-11-21:
The newer driver build power on failed .. v1.99.0-271-g790e82a

- [x] Journal log 
[barreleye_v1.99.0-271-g790e82a_journal.txt](https://github.com/openbmc/openbmc/files/602846/barreleye_v1.99.0-271-g790e82a_journal.txt)

geissonator, 2016-11-21:
@gkeishin please open new issues for things like this.  It's confusing to track multiple problems using the same issue.  I think the original problem you opened this issue for was resolved.  This new issue may be related to the PCIe workaround we brought in, I see this during power on:

```
Oct 20 20:00:14 barreleye settings_manager.py[829]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Oct 20 20:00:14 barreleye settings_manager.py[829]: Traceback (most recent call last):
Oct 20 20:00:14 barreleye settings_manager.py[829]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Oct 20 20:00:15 barreleye settings_manager.py[829]:     self._handler(*args, **kwargs)
Oct 20 20:00:15 barreleye settings_manager.py[829]: TypeError: settings_signal_handler() takes exactly 5 arguments (4 given)
Oct 20 20:00:15 barreleye pcie_slot_present.exe[880]: Checking Presence: SLOT0_RISER_PRESENT
```

gkeishin, 2016-11-21:
Sure @geissonator Open for the new one https://github.com/openbmc/openbmc/issues/824

So we can close this current one ?

geissonator, 2016-11-21:
Thanks, yep this was fixed under #806

closed 2016-11-21 by geissonator