openbmc/openbmc #802: led: Move all existing code to new LED interfaces

geissonator, 2016-11-11:
This is a follow on to #736  and #737 .  This is to go through all code using the skeleton LED interfaces and move them to the new one's defined and implemented in 736 and 737.

vishwabmc, 2016-12-01:
- [ ] Search openbmc# and find the instances
- [ ] If the code is in production, make the changes
- [ ] Test the changes
- [ ] If the code is in Automation, help FTC team do the changes.

vishwabmc, 2017-02-02:
So far, have found these instances : 
```
skeleton chassisctl  -- Asked Andrew on what his plans are on these .

102    @dbus.service.method(DBUS_NAME,
103                         in_signature='', out_signature='')
104    def setIdentify(self):
105        print "Turn on identify"
106        intf = self.getInterface('identify_led')
107        intf.setOn()
108        return None
109
110    @dbus.service.method(DBUS_NAME,
111                         in_signature='', out_signature='')
112    def clearIdentify(self):
113        print "Turn on identify"
114        intf = self.getInterface('identify_led')
115        intf.setOff()
116        return None```

These are the other ones that Brad gave me

`meta-phosphor/common/recipes-phosphor/leds/obmc-control-led`
```./meta-openpower/meta-ibm/meta-palmetto/recipes-phosphor/leds/obmc-control-led%.bbappend
./meta-openpower/meta-rackspace/meta-barreleye/recipes-phosphor/leds/obmc-control-led%.bbappend```

vishwabmc, 2017-02-02:
Couple things to be done here as I was going through.

1/. Making the above unit file changes is easy.

2/ Also need to see why the above targets are written only for Palmetto and Barreleye and not any open-power systems.

2/. What also needs to be written is the YAML definition file for the openpower systems describing the LED layout. I see that Baralleleye has just 3 leds.

vishwabmc, 2017-02-02:
@spinler Where do I get the LED definition for these non MRW systems ?. I did not see them under meta-openpower/..../mrw/..

spinler, 2017-02-02:
Are they just in the <System>.py files in skeleton?

vishwabmc, 2017-03-23:
/lib/systemd/system/obmc-power-start@0.target.wants/

lrwxrwxrwx    1 root     root            32 Mar 22  2017 obmc-led-group-start@PowerOn.service -> ../obmc-led-group-start@.service
lrwxrwxrwx    1 root     root            26 Mar 22  2017 obmc-led-start@identify.service -> ../obmc-led-start@.service
lrwxrwxrwx    1 root     root            24 Mar 22  2017 obmc-max-fans.service -> ../obmc-max-fans.service
root@barreleye:/lib/systemd/system/obmc-power-start@0.target.wants# 

I can see that The Asserted State of PowerOn 

root@barreleye:/lib/systemd/system/obmc-power-start@0.target.wants# dbus-send --system --print-reply --dest=xyz.openbmc_project.LED.GroupManager /xyz/openbmc_project/led/groups/BmcBooted org.freedesktop.DBus.Properties.Get string:"xyz.openbmc_project.Led.Group" string:"Asserted"
method return time=1490065776.223761 sender=:1.3 -> destination=:1.173 serial=56 reply_serial=2
   variant       boolean true

root@barreleye:/lib/systemd/system/obmc-power-start@0.target.wants# dbus-send --system --print-reply --dest=xyz.openbmc_project.LED.Controller.heartbeat /xyz/openbmc_project/led/physical/heartbeat org.freedesktop.DBus.Properties.Get string:"xyz.openbmc_project.Led.Physical" string:"State"
method return time=1490065900.076492 sender=:1.14 -> destination=:1.177 serial=12 reply_serial=2
   variant       string "xyz.openbmc_project.Led.Physical.Action.On"

BMC booted Group and the Physical LED turned ON as part of BMC_READY

```
root@barreleye:/lib/systemd/system/obmc-power-start@0.target.wants# dbus-send --system --print-reply --dest=xyz.openbmc_project.LED.GroupManager /xyz/openbmc_project/led/groups/PowerOn org.freedesktop.DBus.Properties.Get string:"xyz.openbmc_project.Led.Group" string:"Asserted"method return time=1490065761.926525 sender=:1.3 -> destination=:1.171 serial=55 reply_serial=2
   variant       boolean true

root@barreleye:/lib/systemd/system/obmc-power-start@0.target.wants# dbus-send --system --print-reply --dest=xyz.openbmc_project.LED.Controller.beep /xyz/openbmc_project/led/physical/beep org.freedesktop.DBus.Properties.Get string:"xyz.openbmc_project.Led.Physical" string:"State"
method return time=1490065871.311710 sender=:1.13 -> destination=:1.175 serial=23 reply_serial=2
   variant       string "xyz.openbmc_project.Led.Physical.Action.On"

PowerOn Group and the Physical LED turned ON as part of PowerOn

root@barreleye:/lib/systemd/system/obmc-power-start@0.target.wants# 
```

vishwabmc, 2017-03-23:
Since the PowerOn and BmcBooted group is working now, just need to remove the old target since it ends up lighting up other LED also.

1/. Remove the old obmc-led-start@,service

============

Since I am making this change anyway, I can add a PowerOn group to Palmetto in the bbappend.

vishwabmc, 2017-03-23:
I flashed a Palmetto with this CI and made sure that needed LEDs were turned On / Off.

closed 2017-03-31 by rfrandse