openbmc/openbmc #803: led: Generate required LED information from the MRW

geissonator, 2016-11-11:
Ties in with #737 and #736 - this is to make the LED config determination data driven from the MRW.

@spinler will want to provide some input.

vishwabmc, 2016-12-01:
- [ ] Understand basic PERL
- [ ] Understand how MRW is structured
- [ ] Understand Norm's Target.pm
- [ ] Work with Matt to get Witherspoon XML having LED targets in it.
- [ ] Write the perl script to generate the YAML file 

Note : I am not very sure if we have all the information available. If something is missing, we need to add that into MRW.

vishwabmc, 2016-12-13:
https://gerrit.openbmc-project.xyz/1488

geissonator, 2017-01-17:
Need recipe for this script and a recipe to feed the MRW data to the script

closed 2017-02-01 by geissonator