openbmc/openbmc #804: Power state remains 0 after starting power on.

micwalsh, 2016-11-11:
Starting state was:
state:
  state[power_state]:                             0
  state[bmc_state]:                               HOST_POWERED_OFF
  state[boot_progress]:                           Off

Initiated power on...
URI:https://beye6/org/openbmc/control/chassis0//action/powerOn, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}

After 3 mins the power_stae was still off, even though bmc_state had changed:
state:
  state[power_state]:                             0
  state[bmc_state]:                               HOST_POWERING_ON
  state[boot_progress]:                           Off

Prior to this issue, I had successfully run 8 'BMC Power On" and 8 "BMC Power Off" tests.

[20161111150532297875_ffdc_report.txt](https://github.com/openbmc/openbmc/files/586795/20161111150532297875_ffdc_report.txt)

==========================================================================================
Copy this data to the defect:

openbmc_nickname:                                 bard-04
openbmc_host:                                     beye6
openbmc_username:                                 root
openbmc_password:                                 0penBmc
os_host:                                          beye7
os_username:                                      root
os_password:                                      P@ssw0rd
pdu_host:                                         rmtpdu17
pdu_username:                                     admin
pdu_password:                                     admin
pdu_slot_no:                                      1
openbmc_serial_host:                              Ran1
openbmc_serial_port:                              2003
boot_stack:                                       
boot_list:                                        BMC Power On:BMC Power Off
max_num_tests:                                    600
plug_in_dir_paths:                                /afs/rch/projects/esw/dvt/autoipl/apollodev/bin/plug_ins/DB_Logging
status_file_path:                                 /gsa/ausgsa/projects/a/autoipl/status/bard-04.161111.115858.status
openbmc_model:                                    barreleye
boot_pass:                                        16
boot_fail:                                        1
test_mode:                                        0
quiet:                                            0
debug:                                            0
------------------------------------------------------------------------------------------
Last 10 boots:

#(CST) 2016/11/11 12:33:45.316967 -    0.064574 - Doing "BMC Power On".
------------------------------------------------------------------------------------------

FFDC data files:
/gsa/ausgsa/projects/a/autoipl/FFDC/20161111123705987729_ObmcBootTest/20161111123705987729_RandomizedBootTesting/20161111123705987729_BMC_dmesg
/gsa/ausgsa/projects/a/autoipl/FFDC/20161111123705987729_ObmcBootTest/20161111123705987729_RandomizedBootTesting/20161111123705987729_BMC_journalctl.log
/gsa/ausgsa/projects/a/autoipl/FFDC/20161111123705987729_ObmcBootTest/20161111123705987729_RandomizedBootTesting/20161111123705987729_BMC_proc_list
/gsa/ausgsa/projects/a/autoipl/FFDC/20161111123705987729_ObmcBootTest/20161111123705987729_RandomizedBootTesting/20161111123705987729_ffdc_report.txt

==========================================================================================

micwalsh, 2016-11-11:
cat /etc/os-release
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=0.1.0
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=20160914141849

closed 2016-11-11 by micwalsh

amboar, 2016-11-14:
@micwalsh Even though you have closed the issue can you please expand on what the problem turned out to be, and what the solution was (if any)? If someone else sees similar symptoms this information will be helpful.

micwalsh, 2016-11-14:
I realized I had a rather old build loaded on the machine.  I'll be repeating this test very soon with a newer build.  If it re-creates, I'll re-open this issue.