openbmc/openbmc #806: [Build v1.99.0-246-g6026e60 ] Phosphor settings services are not running

gkeishin, 2016-11-14:
The Host services settings /org/openbmc/settings are not up on the new build on all systems.
- [x] Journal Log 
[barreleye_v1.99.0-246-g6026e60_journal.txt](https://github.com/openbmc/openbmc/files/588990/barreleye_v1.99.0-246-g6026e60_journal.txt)

   
GET https://xx.xx.xx.xx/org/openbmc/
{
  "data": [
    "/org/openbmc/control", 
    "/org/openbmc/managers", 
    "/org/openbmc/HostServices", 
    "/org/openbmc/UserManager", 
    "/org/openbmc/buttons", 
    "/org/openbmc/records", 
    "/org/openbmc/inventory", 
    "/org/openbmc/NetworkManager", 
    "/org/openbmc/watchdog", 
    "/org/openbmc/sensors", 
    "/org/openbmc/HostIpmi"
  ], 
  "message": "200 OK", 
  "status": "ok"
}

```
Oct 20 19:58:30 barreleye systemd[1]: org.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE
Oct 20 19:58:30 barreleye systemd[1]: org.openbmc.settings.Host.service: Unit entered failed state.
Oct 20 19:58:30 barreleye systemd[1]: org.openbmc.settings.Host.service: Failed with result 'exit-code'.
Oct 20 19:58:30 barreleye systemd[1]: org.openbmc.settings.Host.service: Service hold-off time over, scheduling restart.
Oct 20 19:58:44 barreleye settings_manager.py[722]: Running HostSettingsService
Oct 20 19:58:44 barreleye systemd[1]: Starting Wait for /org/openbmc/settings/host0...
```

geissonator, 2016-11-14:
I made a comment in https://gerrit.openbmc-project.xyz/#/c/719/11/settings_manager.py@44   I think it's a simple fix, otherwise we should just revert the change in question until we can get things fixed.

geissonator, 2016-11-14:
Potential fix from Sergey - https://gerrit.openbmc-project.xyz/#/c/1172/

closed 2016-11-15 by williamspatrick