openbmc/openbmc #808: Console client connection fails when streaming a large amount of data

dbirk, 2016-11-14:
When using obmc-console-client, if I boot a vm with PFW 'cidebug?' on, which produces a large amount of console text as the partitioon boot, eventually the console connection fails with '~Connection closed' and control reverts back to the BMC prompt. Easily recreated, can probably do it every time I try. Someone can contact me directly if they need further details.

causten, 2016-11-14:
the official way to get the console is to ssh -p 2200 <bmc ip>

Try that, if it still can't hold up then further investigation is warranted.  That obmc-console-client is a dev tool that was around before we got the correct ssh technique

dbirk, 2016-11-14:
Ok port 2200 worked just fine, thanks, closing issue.

closed 2016-11-14 by dbirk