openbmc/openbmc #809: Host watchdog gets enabled on debug mode

anoo1, 2016-11-14:
We've regressed and the host watchdog is now enabled when the BMC is set to debug (cronus) mode, the watchdog should not be enabled in this mode. This hampers P9 debug as the system powers off automatically due to the watchdog timer expiring.
Since the skeleton repo is planned to be refactor, maybe fixing the watchdog can be done as part of the state management (power on/off) interface refactoring.

anoo1, 2016-11-14:
This was seen on firestone on an 'older' (~2 months back) build, need to check the latest master on a supported platform like barreleye.

bradbishop, 2016-11-17:
For debug mode I was thinking we could just start the power on target:

systemctl start obmc-power-on@0.service

That might not be the right unit name.

anoo1, 2017-09-19:
As Brad mentioned, just need to start the power on target, it may be called "chassis on" nowadays. Closing.

closed 2017-09-19 by anoo1