openbmc/openbmc #810: Don't Build Image

Youngzhong, 2016-11-15:
When we build image to use "bitbake obmc-phosphor-image" on 1.0 code,We don't  find 16M image in openbmc-master\build\tmp\deploy\images. Why? We need change config,when we build image.Thanks!

Youngzhong, 2016-11-15:
Dear All
We build the image that it need support AST2400.We need change config in" openbmc-master\build\conf\local.conf"? and where?

williamspatrick, 2016-11-15:
We do not support 16M images, only 32M or above.

Depending on which machine you are trying to build for you need to pre-set some environment variables.  Here is an example for the "Palmetto" machine, but you can change the TEMPLATECONF and MACHINE variables to be any other machine you find under meta-openbmc-machines.
https://github.com/openbmc/docs/blob/master/cheatsheet.md#building-for-palmetto

Youngzhong, 2016-11-15:
Dear william
Got it.I will try it.Thanks.I will change the IC from 16M to 32M.

Youngzhong, 2016-11-15:
Dear William
This is failure info。I try 3 times. The failure is the same . Is it a question of my network speed?
===================================================================
kunlun@kunlun-550-171cn:~/Desktop/openbmc-master/build$ bitbake obmc-phosphor-image
Loading cache: 100% |###########################################| ETA:  00:00:00
Loaded 2480 entries from dependency cache.
Parsing recipes: 100% |#########################################| Time: 00:00:00
Parsing of 1918 .bb files complete (1917 cached, 1 parsed). 2481 targets, 378 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "i686-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-openbmc-linux-gnueabi"
MACHINE           = "barreleye"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta              
meta-poky         
meta-oe           
meta-networking   
meta-python       
meta-virtualization 
meta-phosphor     
meta-aspeed       
meta-ast2400      
meta-openpower    
meta-rackspace    
meta-barreleye    = "<unknown>:<unknown>"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: u-boot-mkimage-native-v2016.07+gitAUTOINC+f9b2a49f57-r0 do_fetch: Failed to fetch URL git://git@github.com/openbmc/u-boot.git;branch=v2016.07-aspeed-openbmc;protocol=https, attempting MIRRORS if available
ERROR: u-boot-mkimage-native-v2016.07+gitAUTOINC+f9b2a49f57-r0 do_fetch: Fetcher failure: Fetch command failed with exit code 128, output:
Cloning into bare repository '/home/kunlun/Desktop/openbmc-master/build/downloads/git2/github.com.openbmc.u-boot.git'...
error: RPC failed; result=56, HTTP code = 200
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

ERROR: u-boot-mkimage-native-v2016.07+gitAUTOINC+f9b2a49f57-r0 do_fetch: Function failed: Fetcher failure for URL: 'git://git@github.com/openbmc/u-boot.git;branch=v2016.07-aspeed-openbmc;protocol=https'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/kunlun/Desktop/openbmc-master/build/tmp/work/i686-linux/u-boot-mkimage-native/v2016.07+gitAUTOINC+f9b2a49f57-r0/temp/log.do_fetch.20645
ERROR: Task 199 (virtual:native:/home/kunlun/Desktop/openbmc-master/meta-phosphor/common/recipes-bsp/u-boot/u-boot-mkimage_2016.07.bb, do_fetch) failed with exit code '1'
WARNING: u-boot-fw-utils-v2016.07+gitAUTOINC+f9b2a49f57-r0 do_fetch: Failed to fetch URL git://git@github.com/openbmc/u-boot.git;branch=v2016.07-aspeed-openbmc;protocol=https, attempting MIRRORS if available
ERROR: u-boot-fw-utils-v2016.07+gitAUTOINC+f9b2a49f57-r0 do_fetch: Fetcher failure: Fetch command failed with exit code 128, output:
Cloning into bare repository '/home/kunlun/Desktop/openbmc-master/build/downloads/git2/github.com.openbmc.u-boot.git'...
error: RPC failed; result=56, HTTP code = 200
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

ERROR: u-boot-fw-utils-v2016.07+gitAUTOINC+f9b2a49f57-r0 do_fetch: Function failed: Fetcher failure for URL: 'git://git@github.com/openbmc/u-boot.git;branch=v2016.07-aspeed-openbmc;protocol=https'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/kunlun/Desktop/openbmc-master/build/tmp/work/barreleye-openbmc-linux-gnueabi/u-boot-fw-utils/v2016.07+gitAUTOINC+f9b2a49f57-r0/temp/log.do_fetch.20656
ERROR: Task 362 (/home/kunlun/Desktop/openbmc-master/meta-phosphor/common/recipes-bsp/u-boot/u-boot-fw-utils_2016.07.bb, do_fetch) failed with exit code '1'
WARNING: u-boot-v2016.07+gitAUTOINC+f9b2a49f57-r0 do_fetch: Failed to fetch URL git://git@github.com/openbmc/u-boot.git;branch=v2016.07-aspeed-openbmc;protocol=https, attempting MIRRORS if available
ERROR: u-boot-v2016.07+gitAUTOINC+f9b2a49f57-r0 do_fetch: Fetcher failure: Fetch command failed with exit code 128, output:
Cloning into bare repository '/home/kunlun/Desktop/openbmc-master/build/downloads/git2/github.com.openbmc.u-boot.git'...
error: RPC failed; result=56, HTTP code = 200
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

ERROR: u-boot-v2016.07+gitAUTOINC+f9b2a49f57-r0 do_fetch: Function failed: Fetcher failure for URL: 'git://git@github.com/openbmc/u-boot.git;branch=v2016.07-aspeed-openbmc;protocol=https'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/kunlun/Desktop/openbmc-master/build/tmp/work/barreleye-openbmc-linux-gnueabi/u-boot/v2016.07+gitAUTOINC+f9b2a49f57-r0/temp/log.do_fetch.20673
ERROR: Task 77 (/home/kunlun/Desktop/openbmc-master/meta-phosphor/common/recipes-bsp/u-boot/u-boot_2016.07.bb, do_fetch) failed with exit code '1'
===================================================================

williamspatrick, 2016-11-15:
@shenki , @bradbishop - I know we haven't had any problems lately building, but the SRC_URI looks funny.  Any ideas?

```
SRC_URI = "git://git@github.com/openbmc/u-boot.git;branch=${UBRANCH};protocol=https"
```

williamspatrick, 2016-11-15:
I'm seeing reports from some of our internal build tools that Github was having issues earlier today, even thought their status doesn't seem to report as such. @Youngzhong can you retry and see if issues have been resolved?

amboar, 2016-11-15:
@williamspatrick How does the SRC_URI look funny? It looks roughly like what I'd expect it to.

shenki, 2016-11-15:
It has `protocol=https` but the URI is `git://`, which is not HTTPS

amboar, 2016-11-16:
@shenki That's [fine according to the documentation](https://www.yoctoproject.org/docs/1.6/bitbake-user-manual/bitbake-user-manual.html#git-fetcher):

> "protocol": The protocol used to fetch the files. The default is "git" when a hostname is set. If a hostname is not set, the Git protocol is "file". You can also use "http", "https", "ssh" and "rsync". 

and

> Here are some example URLs:
>
>    SRC_URI = "git://git.oe.handhelds.org/git/vip.git;tag=version-1"
>    SRC_URI = "git://git.oe.handhelds.org/git/vip.git;protocol=http"

williamspatrick, 2016-11-16:
The `git://git@` plus `protocol=https` is specifically what had me confused.  'git@github.com' is the ssh user for github access.  So we have 3 different protocols all partially identified here.

amboar, 2016-11-16:
@williamspatrick yeah, supplying a user for unauthenticated https is odd, but in my mind shouldn't break. I wouldn't be surprised if I'm wrong though.

Youngzhong, 2016-11-16:
Dear William,Shenki and Amboar
Thank you very much for your support. I have already built 32M image.and Now I am testing.May be the problem of my speed? I try 5 times. Happy,The fifth times is OK.

Youngzhong, 2016-11-16:
Dear William,Shenki and Amboar
The following method is correct?
1.Support AST2400
$ cd openbmc
$ TEMPLATECONF=openbmc-master\meta-openbmc-bsp\meta-aspeed\meta-ast2400\conf . openbmc-env
$ bitbake obmc-phosphor-image

2.Support AST2500
$ cd openbmc
$ TEMPLATECONF=openbmc-master\meta-openbmc-bsp\meta-aspeed\meta-ast2500\conf. openbmc-env
$ bitbake obmc-phosphor-image

3.Support PowerPC
$ cd openbmc
$ TEMPLATECONF=openbmc-master\meta-openbmc-machines\meta-openpower\meta-rackspace\meta-barreleye\conf. openbmc-env
$ bitbake obmc-phosphor-image

williamspatrick, 2016-11-16:
'ast2400' and 'ast2500' will compile an image for a raw Aspeed evaulation board.  These do not have as many of the OpenBMC userspace applications installed because many of those are for managing a server.

'barreleye' is not a generic "Support Power" but a specific machine.  Each machine has its own templateconf.  So you need to make sure to build for the machine you are trying to put an image onto.  'Palmetto', "Barreleye' are the two best supported Power8 machines.

Youngzhong, 2016-11-22:
Dear All
This is ok.Thanks!

closed 2016-11-22 by Youngzhong