openbmc/openbmc #811: [Barreleye v1.99.0-251-gb78c6de ] Powercap properties not loaded

gkeishin, 2016-11-16:
` 
root@barreleye:~# obmcutil state
 = HOST_BOOTED
root@barreleye:~#
`

- [x] Journal Logs  
[barreleye_v1.99.0-251-gb78c6de_journal.txt](https://github.com/openbmc/openbmc/files/595228/barreleye_v1.99.0-251-gb78c6de_journal.txt)

```
rango@ubuntu:~/openbmc-test-automation$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET https://xx.xx.xx.xx/org/openbmc/sensors/
{
  "data": [
    "/org/openbmc/sensors/HSCB", 
    "/org/openbmc/sensors/HSCA", 
    "/org/openbmc/sensors/temperature", 
    "/org/openbmc/sensors/tach", 
    "/org/openbmc/sensors/host", 
    "/org/openbmc/sensors/voltage", 
    "/org/openbmc/sensors/speed"
  ], 
  "message": "200 OK", 
  "status": "ok"
}rango@ubuntu:~/openbmc-test-automation$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET https://xx.xx.xx.xx/org/openbmc/sensors/host/
{
  "data": [
    "/org/openbmc/sensors/host/powercap", 
    "/org/openbmc/sensors/host/OperatingSystemStatus", 
    "/org/openbmc/sensors/host/cpu0", 
    "/org/openbmc/sensors/host/cpu1", 
    "/org/openbmc/sensors/host/BootCount", 
    "/org/openbmc/sensors/host/BootProgress"
  ], 
  "message": "200 OK", 
  "status": "ok"
}rango@ubuntu:~/openbmc-test-automation$ 
```

Expecting these properties under sensors/ 

/org/openbmc/sensors/powercap/curr_cap
/org/openbmc/sensors/powercap/max_cap
/org/openbmc/sensors/powercap/min_cap
/org/openbmc/sensors/powercap/system_power
/org/openbmc/sensors/powercap/user_cap
/org/openbmc/sensors/host/powercap/action/setValue

ratagupt, 2016-11-17:
https://github.com/openbmc/openbmc/issues/557 
is it related to this one?

gkeishin, 2016-11-17:
@ratagupt Yes and no.. 
- [x] Yes because we have this concern that there are two places one in Settings and another in Sensors..
- [x] No because we weren't expecting this interface to go off like this, probably if move to other URI where is that now ?

gkeishin, 2016-11-29:
Looks like the path exist just that i doesn't show up the properties on query

`}rango@ubuntu:~/openbmc-test-automation$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET https://xx.xx.xx.xx/org/openbmc/sensors/powercap/`
{
  "data": [
    "/org/openbmc/sensors/powercap/system_power", 
    "/org/openbmc/sensors/powercap/curr_cap", 
    "/org/openbmc/sensors/powercap/min_cap", 
    "/org/openbmc/sensors/powercap/max_cap", 
    "/org/openbmc/sensors/powercap/user_cap", 
    "/org/openbmc/sensors/powercap/n_cap"
  ], 
  "message": "200 OK", 
  "status": "ok"
}
`rango@ubuntu:~/openbmc-test-automation$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET https://xx.xx.xx.xx/org/openbmc/sensors`
{
  "data": {}, 
  "message": "200 OK", 
  "status": "ok"
}rango@ubuntu:~/openbmc-test-automation$

gkeishin, 2016-11-29:
I see that the OCC min power cap is 1100  and max is 1200 watts ? is this the default setting value.. earlier it use to be ranging from 0 to 1000 watts... anything new changed since ?

rahulmah, 2016-12-06:
@gkeishin : I believe this issue can be closed. I dont see now the problem while power cap properties loading. Tested and checked with v1.99.0-309 build

closed 2017-03-31 by gkeishin