openbmc/openbmc #812: Internal server error on REST request

shenki, 2016-11-17:
I have a util that does a GET on `/org/openbmc/inventory/system`. With build [362](https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=palmetto/362/artifact/images/palmetto/)

v1.99.0-257-g85b85b3b6ba4 85b85b3b6ba406

```
$ obmc-ctrl --bmc 10.104.0.96 version
{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException('busy processing bus traffic',)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 858, in _handle", 
      "    route, args = self.router.match(environ)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 740, in custom_router_match", 
      "    route.callback._setup(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 80, in _setup", 
      "    return self.setup(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 360, in setup", 
      "    request.route_data['map'] = self.find(path, callback)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 351, in find", 
      "    path=path)}", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 97, in try_mapper_call", 
      "    return f(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 55, in get_object", 
      "    retries)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 42, in retry", 
      "    return func()", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 54, in <lambda>", 
      "    lambda: self.iface.GetObject(path),", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic"
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
}
``

shenki, 2016-11-17:
Trying again I got a valid response. I assume that the system was still starting up.

We should return a better error message.

ratagupt, 2016-11-17:
As the system is booting up and during that time there would be a more dbus traffic,Can we start the rest server once BMC reaches to BMC_READY state...Or Rest server should be the last one to start before BMC reaches to BMC_READY state?
@Brad:- what is your thought in this?

geissonator, 2016-11-17:
I'm seeing a similar error on boots of the latest code.  This is from the journal, there was no external stimuli at the time.  Let me know if we need a separate issue for this.  Looks like a generic issue where DBUS is getting bogged down and we don't have any internal retries?

```
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-257"
VERSION_ID="v1.99.0-257-g85b85b3-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-257"
BUILD_ID="v1.99.0"
```

```
Oct 20 19:59:18 barreleye settings_manager.py[804]: Traceback (most recent call last):
Oct 20 19:59:18 barreleye settings_manager.py[804]:   File "/usr/sbin/settings_manager.py", line 245, in <module>
Oct 20 19:59:18 barreleye settings_manager.py[804]:     allobjects = create_object(s.SETTINGS)
Oct 20 19:59:18 barreleye settings_manager.py[804]:   File "/usr/sbin/settings_manager.py", line 71, in create_object
Oct 20 19:59:18 barreleye settings_manager.py[804]:     paths = mapper.get_subtree_paths(setting['subtree'], 0)
Oct 20 19:59:18 barreleye settings_manager.py[804]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 60, in get_subtree_paths
Oct 20 19:59:18 barreleye settings_manager.py[804]:     retries)
Oct 20 19:59:18 barreleye settings_manager.py[804]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 42, in retry
Oct 20 19:59:18 barreleye settings_manager.py[804]:     return func()
Oct 20 19:59:18 barreleye settings_manager.py[804]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 59, in <lambda>
Oct 20 19:59:18 barreleye settings_manager.py[804]:     lambda: self.iface.GetSubTreePaths(path, depth),
Oct 20 19:59:18 barreleye settings_manager.py[804]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Oct 20 19:59:18 barreleye settings_manager.py[804]:     **keywords)
Oct 20 19:59:18 barreleye settings_manager.py[804]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Oct 20 19:59:18 barreleye settings_manager.py[804]:     message, timeout)
Oct 20 19:59:18 barreleye settings_manager.py[804]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic
Oct 20 19:59:18 barreleye systemd[1]: org.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE
Oct 20 19:59:18 barreleye systemd[1]: Failed to start Phosphor Settings Manager.
Oct 20 19:59:18 barreleye systemd[1]: org.openbmc.settings.Host.service: Unit entered failed state.
Oct 20 19:59:18 barreleye systemd[1]: org.openbmc.settings.Host.service: Failed with result 'exit-code'.
```

geissonator, 2016-11-17:
Sometimes we seem to recover automatically ok, but sometimes I have to restart the system manager like in this case:

It was just sitting in the BASE_APPS state until I restarted it.

```
[barreleye]# systemctl status org.openbmc.managers.System.service | cat
\ufffd\ufffdorg.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2016-10-20 19:58:39 UTC; 9min ago
 Main PID: 729 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           \ufffd\ufffd\ufffd\ufffd\ufffd\ufffd729 python /usr/sbin/system_manager.py

Oct 20 19:58:58 barreleye system_manager.py[729]: Traceback (most recent call last):
Oct 20 19:58:58 barreleye system_manager.py[729]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Oct 20 19:58:58 barreleye system_manager.py[729]:     *message.get_args_list()))
Oct 20 19:58:58 barreleye system_manager.py[729]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 91, in <lambda>
Oct 20 19:58:58 barreleye system_manager.py[729]:     path, retry, x))
Oct 20 19:58:58 barreleye system_manager.py[729]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 103, in get_object_error
Oct 20 19:58:58 barreleye system_manager.py[729]:     self.error_callback(e)
Oct 20 19:58:58 barreleye system_manager.py[729]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 56, in default_error
Oct 20 19:58:58 barreleye system_manager.py[729]:     raise e
Oct 20 19:58:58 barreleye system_manager.py[729]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic
[barreleye]# systemctl restart org.openbmc.managers.System.service  
[barreleye]# state
 = BMC_READY
```

geissonator, 2016-11-18:
Ratan, Patrick was thinking we want to look at "dbus-monitor —system" during the fail time to see what's happening on the dbus.  His thought is maybe the mapper code is doing a hard poll on the bus.  Brad may be able to help with some ideas.

bradbishop, 2016-11-28:
There are probably two different problems here.

At startup, the bus is very busy as applications start and I expect the mapper to be busy.  I agree that when the mapper is busy, the REST server should return a 503.

Getting a busy once we've reached ready state is likely a different problem.  Once all the applications are up, I don't expect a busy.  Lets tackle that problem in a different issue.

bradbishop, 2016-11-28:
@ratagupt fyi...the rest server isn't started until someone makes a query.

bradbishop, 2016-11-28:
https://gerrit.openbmc-project.xyz/#/c/1319/

closed 2016-11-29 by williamspatrick