openbmc/openbmc #814: GPIO abstraction for host-control

mine260309, 2016-11-17:
Following the [commit](https://github.com/openbmc/skeleton/commit/20a194169c527fbded95749acd29ef44dfbe70c7) and #315, GPIOs used in host-control shall be abstracted as well to make the code independent from the GPIO names, otherwise the c code will have to deal with the differences of GPIOs between P8 and P9.

This task is to address the issue, using the similar solution to abstract host-control related GPIOs.

mine260309, 2016-11-17:
Take Romulus as example, the previous `POWER_CONFIG` will be renamed and changed to something like below:
```
GPIO_CONFIGS = {
    'power_config' : {
        'power_good_in' : 'SYS_PWROK_BUFF',
        'power_up_outs' : [
            ('SOFTWARE_PGOOD', True),
            ('BMC_POWER_UP', True),
        ],
        'reset_outs' : [
            ('BMC_CP0_RESET_N', False),
            ('BMC_CP0_PERST_ENABLE_R', False),
        ],
    },
    'hostctl_config' : {
        'fsi_data' : 'FSI_DATA',
        'fsi_clk' : 'FSI_CLK',
        'fsi_enable' : 'FSI_ENABLE',
        'cronus_sel' : 'CRONUS_SEL',
        'optionals' : [
            ('BMC_THROTTLE', True),
            ('IDBTN', False), # TODO: remove these two after optionals are implemented in control_host_obj.c
        ],
    },
}
```

And in `pysystemmgr/system_manager.py`, the function `getPowerConfiguration()` will be renamed to `getGpioConfiguration()` with updated signature.

Welcome to suggest better names for `GPIO_CONFIGS` and `getPowerConfiguration()`

mine260309, 2016-11-21:
Submitted review:
https://gerrit.openbmc-project.xyz/#/c/1248
https://gerrit.openbmc-project.xyz/#/c/1249

The changes are tested on Romulus and Palmetto, and confirmed that power_control and host_control are working OK.

amboar, 2016-11-22:
@shenki should we try to incorporate some of your thoughts on GPIO handling into @mine260309's changes here? Might be a useful proving-ground.

closed 2017-01-12 by williamspatrick