openbmc/openbmc #815: Network interface dropping once we get to BMC_READY

geissonator, 2016-11-17:
I've been seeing this for a while (ever since we got the new network code in maybe?) but I've been mostly ignoring it since it didn't seem to affect anything.   For whatever reason, that changed in our recent code it seems.

The code update flow is to do a PrepareForUpdate REST call, which reboots the BMC.  It then waits for BMC_READY and starts to scp the flash image over.  During that scp, the network interface seems to reconfigure itself and interrupt the scp.  My ssh session into the system gets a bit of a hiccup but stays connected, the scp, and sometimes other REST calls are not so resilient.

```
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-257"
VERSION_ID="v1.99.0-257-g85b85b3-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-257"
BUILD_ID="v1.99.0"
```

Here's the relevant data from the journal starting with us at BMC_READY and getting what appears to be a network reconfig:

```
Oct 20 19:59:19 barreleye system_manager.py[789]: SystemManager Goto System State: BMC_READY
Oct 20 19:59:19 barreleye system_manager.py[789]: Running System State: BMC_READY
Oct 20 19:59:20 barreleye phosphor-gevent[762]: ::c043:aab6:8e91:f402%1326025032 - - [2016-10-20 19:59:20] "POST /org/openbmc/managers/System//action/getSystemState HTTP/1.1" 200 176 1.308694
Oct 20 19:59:20 barreleye phosphor-gevent[762]: ::c043:aab6:8e91:f402%1326025032 - - [2016-10-20 19:59:20] "POST /login HTTP/1.1" 200 320 0.037372
<inventory stuff>
Oct 20 19:59:22 barreleye phosphor-gevent[762]: ::c043:aab6:8e91:f402%1326025032 - - [2016-10-20 19:59:22] "PUT /org/openbmc/control/flash/bmc/attr/preserve_network_settings HTTP/1.1" 200 169 1.672098
Oct 20 19:59:22 barreleye phosphor-gevent[762]: ::c043:aab6:8e91:f402%1326025032 - - [2016-10-20 19:59:22] "POST /login HTTP/1.1" 200 320 0.037840
Oct 20 19:59:22 barreleye phosphor-gevent[762]: ::c043:aab6:8e91:f402%1326025032 - - [2016-10-20 19:59:22] "GET /org/openbmc/control/flash/bmc HTTP/1.1" 200 380 0.126098
Oct 20 19:59:22 barreleye systemd[1]: Started SSH Per-Connection Server (9.41.165.94:59406).
Oct 20 19:59:22 barreleye dropbear[888]: Failed loading /etc/dropbear/dropbear_dss_host_key
Oct 20 19:59:22 barreleye dropbear[888]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Oct 20 19:59:22 barreleye dropbear[888]: Child connection from ::ffff:9.41.165.94:59406
Oct 20 19:59:24 barreleye dropbear[888]: pam_unix(dropbear:account): account root has password changed in future
Oct 20 19:59:24 barreleye dropbear[888]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.94:59406
... <inventory stuff>
Oct 20 19:59:29 barreleye systemd[1]: obmc-read-eeprom@system-chassis-motherboard.service: Main process exited, code=exited, status=1/FAILURE
Oct 20 19:59:29 barreleye systemd[1]: Failed to start Read system/chassis/motherboard EEPROM.
Oct 20 19:59:29 barreleye systemd[1]: obmc-read-eeprom@system-chassis-motherboard.service: Unit entered failed state.
Oct 20 19:59:29 barreleye systemd[1]: obmc-read-eeprom@system-chassis-motherboard.service: Failed with result 'exit-code'.
Oct 20 19:59:29 barreleye systemd[1]: Starting Sync system/chassis/motherboard...
Oct 20 19:59:29 barreleye systemd[1]: Started Wait for /org/openbmc/settings/host0.
Oct 20 19:59:29 barreleye systemd[1]: Reached target Restore Last State.
Oct 20 19:59:29 barreleye systemd[1]: Starting Restore Last State...
Oct 20 19:59:34 barreleye discover_system_state.py[895]: Last System State:0Power Policy:RESTORE_LAST_STATE
Oct 20 19:59:34 barreleye systemd[1]: Started Restore Last State.
Oct 20 19:59:36 barreleye netman.py[728]: Restarting networkd service...
Oct 20 19:59:36 barreleye avahi-daemon[747]: Interface eth0.IPv6 no longer relevant for mDNS.
Oct 20 19:59:36 barreleye systemd-networkd[721]: eth0: Lost carrier
Oct 20 19:59:36 barreleye avahi-daemon[747]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::1e66:6dff:fe85:4ad0.
Oct 20 19:59:36 barreleye avahi-daemon[747]: Interface eth0.IPv4 no longer relevant for mDNS.
Oct 20 19:59:36 barreleye avahi-daemon[747]: Leaving mDNS multicast group on interface eth0.IPv4 with address 9.3.23.30.
Oct 20 19:59:36 barreleye avahi-daemon[747]: Withdrawing address record for fe80::1e66:6dff:fe85:4ad0 on eth0.
Oct 20 19:59:36 barreleye avahi-daemon[747]: Withdrawing address record for 9.3.23.30 on eth0.
Oct 20 19:59:36 barreleye avahi-daemon[747]: Joining mDNS multicast group on interface eth0.IPv4 with address 9.3.23.30.
Oct 20 19:59:36 barreleye systemd-networkd[721]: eth0: Gained carrier
Oct 20 19:59:36 barreleye avahi-daemon[747]: New relevant interface eth0.IPv4 for mDNS.
Oct 20 19:59:36 barreleye avahi-daemon[747]: Registering new address record for 9.3.23.30 on eth0.IPv4.
Oct 20 19:59:36 barreleye systemd-timesyncd[718]: Network configuration changed, trying to establish connection.
Oct 20 19:59:36 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Oct 20 19:59:36 barreleye kernel[724]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Oct 20 19:59:37 barreleye sync_inventory_items.py[892]: Rebooting BMC to set uuid
Oct 20 19:59:37 barreleye systemd[1]: Started Sync system/chassis/motherboard.
Oct 20 19:59:37 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Oct 20 19:59:37 barreleye kernel[724]: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Oct 20 19:59:37 barreleye avahi-daemon[747]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::1e66:6dff:fe85:4ad0.
Oct 20 19:59:37 barreleye avahi-daemon[747]: New relevant interface eth0.IPv6 for mDNS.
Oct 20 19:59:37 barreleye avahi-daemon[747]: Registering new address record for fe80::1e66:6dff:fe85:4ad0 on eth0.*.
Oct 20 19:59:38 barreleye systemd[1]: Stopping Network Service...
Oct 20 19:59:38 barreleye systemd[1]: Stopping Phosphor Webserver...
Oct 20 19:59:38 barreleye systemd[1]: Stopped Network Service.
Oct 20 19:59:38 barreleye systemd[1]: Stopped Phosphor Webserver.
Oct 20 19:59:38 barreleye systemd[1]: Closed Phosphor Webserver socket.
Oct 20 19:59:38 barreleye systemd[1]: Stopping Phosphor Webserver socket.
Oct 20 19:59:38 barreleye systemd[1]: Listening on Phosphor Webserver socket.
Oct 20 19:59:38 barreleye systemd[1]: Started Phosphor Webserver.
Oct 20 19:59:38 barreleye systemd[1]: Starting Network Service...
Oct 20 19:59:38 barreleye systemd-timesyncd[718]: Network configuration changed, trying to establish connection.
Oct 20 19:59:40 barreleye systemd-networkd[908]: eth0: Gained IPv6LL
Oct 20 19:59:40 barreleye systemd-networkd[908]: Enumeration completed
Oct 20 19:59:40 barreleye systemd[1]: Started Network Service.
Oct 20 19:59:40 barreleye systemd[1]: Started Sync system/chassis/io_board.
Oct 20 19:59:40 barreleye systemd[1]: Reached target Phosphor System - Standby.
Oct 20 19:59:40 barreleye systemd[1]: Startup finished in 15.368s (kernel) + 2min 20.775s (userspace) = 2min 36.144s.
Oct 20 19:59:40 barreleye systemd-timesyncd[718]: Network configuration changed, trying to establish connection.
Oct 20 19:59:40 barreleye systemd-timesyncd[718]: Network configuration changed, trying to establish connection.
Oct 20 19:59:41 barreleye avahi-daemon[747]: Withdrawing address record for 9.3.23.30 on eth0.
Oct 20 19:59:41 barreleye avahi-daemon[747]: Leaving mDNS multicast group on interface eth0.IPv4 with address 9.3.23.30.
Oct 20 19:59:41 barreleye avahi-daemon[747]: Interface eth0.IPv4 no longer relevant for mDNS.
Oct 20 19:59:41 barreleye avahi-daemon[747]: Joining mDNS multicast group on interface eth0.IPv4 with address 9.3.23.30.
Oct 20 19:59:41 barreleye avahi-daemon[747]: New relevant interface eth0.IPv4 for mDNS.
Oct 20 19:59:41 barreleye avahi-daemon[747]: Registering new address record for 9.3.23.30 on eth0.IPv4.
Oct 20 19:59:41 barreleye systemd-timesyncd[718]: Network configuration changed, trying to establish connection.
Oct 20 19:59:43 barreleye systemd-timesyncd[718]: Network configuration changed, trying to establish connection.
Oct 20 19:59:45 barreleye systemd-timesyncd[718]: Network configuration changed, trying to establish connection.
Oct 20 19:59:54 barreleye systemd-networkd[908]: eth0: Starting DHCPv6 client after NDisc timeout failed: Invalid argument
Oct 20 19:59:54 barreleye systemd-networkd[908]: eth0: Configured
Oct 20 19:59:54 barreleye systemd-timesyncd[718]: Network configuration changed, trying to establish connection.
```

And here's the partially copied file over that scp connection:
[barreleye]# ls -al /tmp/flashimg 
-rw-r--r--    1 root     root       2260992 Oct 20 19:59 /tmp/flashimg

@gwshan

gwshan, 2016-11-18:
From the log shown as above, the network interface seems brought down and up by application. Please guide who can help checking on it.

Oct 20 19:59:36 barreleye avahi-daemon[747]: New relevant interface eth0.IPv4 for mDNS.
Oct 20 19:59:36 barreleye avahi-daemon[747]: Registering new address record for 9.3.23.30 on eth0.IPv4.
Oct 20 19:59:36 barreleye systemd-timesyncd[718]: Network configuration changed, trying to establish connection.
Oct 20 19:59:36 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Oct 20 19:59:36 barreleye kernel[724]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Oct 20 19:59:37 barreleye sync_inventory_items.py[892]: Rebooting BMC to set uuid

ratagupt, 2016-11-18:
I am still unable to understand how and which application is bringing down the network,Or is it a genuine link failure issue...
There is a network failure, then avahi leaves  the multicast group.and once the link is up then again avahi joins the group..
Oct 20 19:59:36 barreleye netman.py[728]: Restarting networkd service...
Oct 20 19:59:36 barreleye systemd-networkd[721]: eth0: Lost carrier.
Oct 20 19:59:36 barreleye avahi-daemon[747]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::1e66:6dff:fe85:4ad0.
Oct 20 19:59:36 barreleye avahi-daemon[747]: Interface eth0.IPv4 no longer relevant for mDNS.
Oct 20 19:59:36 barreleye avahi-daemon[747]: Leaving mDNS multicast group on interface eth0.IPv4 with address 9.3.23.30.
Oct 20 19:59:36 barreleye systemd-networkd[721]: eth0: Gained carrier..

So here first log suggest that  "Restarting networkd service" happened..
It can happen in two cases 
1) if somehow we called the setAddress4 on network manager
2) If enable the dhcp
(could be through REST/Internal DBUS call or somebody accidentally changed the IP or playing with the dhcp mode. ).

geissonator, 2016-12-05:
Continuing to see this, causing intermittent failures in our automation testing.  Nothing obvious to me on why netman.py is restarting things.

```
Oct 20 19:59:30 barreleye systemd[1]: Starting Sync system/chassis/motherboard...
Oct 20 19:59:31 barreleye phosphor-gevent[766]: ::c073:abb6:8e91:f402%1099860296 - - [2016-10-20 19:59:31] "POST /login HTTP/1.1" 200 320 0.039094
Oct 20 19:59:31 barreleye phosphor-gevent[766]: ::c073:abb6:8e91:f402%1099860296 - - [2016-10-20 19:59:31] "POST /org/openbmc/managers/System//action/getSystemState HTTP/1.1" 200 176 0.270955
Oct 20 19:59:32 barreleye phosphor-gevent[766]: ::c073:abb6:8e91:f402%1099860296 - - [2016-10-20 19:59:32] "POST /login HTTP/1.1" 200 320 0.039127
Oct 20 19:59:33 barreleye netman.py[738]: Restarting networkd service...
Oct 20 19:59:33 barreleye avahi-daemon[734]: Interface eth0.IPv6 no longer relevant for mDNS.
Oct 20 19:59:33 barreleye systemd-networkd[723]: eth0: Lost carrier
Oct 20 19:59:33 barreleye avahi-daemon[734]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::1e66:6dff:fe85:4ad0.
```

```
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-276"
VERSION_ID="v1.99.0-276-g221d855-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-276"
BUILD_ID="v1.99.0"
```

mdmillerii, 2017-02-14:
I notice this in the log
```
Oct 20 19:59:29 barreleye systemd[1]: obmc-read-eeprom@system-chassis-motherboard.service: Main process exited, code=exited, status=1/FAILURE
Oct 20 19:59:29 barreleye systemd[1]: Failed to start Read system/chassis/motherboard EEPROM.
Oct 20 19:59:29 barreleye systemd[1]: obmc-read-eeprom@system-chassis-motherboard.service: Unit entered failed state.
Oct 20 19:59:29 barreleye systemd[1]: obmc-read-eeprom@system-chassis-motherboard.service: Failed with result 'exit-code'.
Oct 20 19:59:29 barreleye systemd[1]: Starting Sync system/chassis/motherboard...
```
Perhaps this lead to the mac being changed and the mac to drop?

geissonator, 2017-06-28:
@gkeishin - we still seeing this one?  Be nice to get rid of that 1 minute sleep in the code update path.

gkeishin, 2017-06-28:
This is more of a Barreleye issue and I haven't seen on Witherspoon so far. So I'll check on Barreleye if this still exists.

gkeishin, 2017-06-28:
I verified on the newer build, this no longer exists.
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-61"
VERSION_ID="v1.99.7-61-gb42df4d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-61"
BUILD_ID="v1.99.7"
root@barreleye:~# uname -a
Linux barreleye 4.10.17-9e95993e2d6edc68af5ca3f2345789e0f450d0db #1 Wed Jun 28 03:18:06 UTC 2017 armv5tejl GNU/Linux
root@barreleye:~#
```

closed 2017-06-28 by gkeishin