openbmc/openbmc #816: [Code update] MTD device partition error for image-bmc

gkeishin, 2016-11-18:
We are trying to code update with the latest image.. looks like the partition is gone for toss

Response code:200, Content:{
  "data": {
    "auto_apply": 0, 
    "clear_persistent_files": 0, 
    "filename": "/tmp/flashimg", 
    "preserve_network_settings": 1, 
    "restore_application_defaults": 0, 
  ```  "status": "Verify error: update: --no-flash --ignore-mount --no-save-files --no-restore-files --no-clean-saved-files\nERROR: Unable to find mtd partiton for image-bmc.\n", ```
    "update_kernel_and_apps": 0
  }, 
  "message": "200 OK", 
  "status": "ok"
}

Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     248232         0    248232   0% /dev
tmpfs                   249508      9068    240440   4% /run
/dev/mtdblock5           14720     14720         0 100% /run/initramfs/ro
/dev/mtdblock6            4096       348      3748   8% /run/initramfs/rw
cow                       4096       348      3748   8% /
tmpfs                   249508         0    249508   0% /dev/shm
tmpfs                   249508         0    249508   0% /sys/fs/cgroup
tmpfs                   249508         0    249508   0% /tmp
tmpfs                   249508         8    249500   0% /var/volatile
root@barreleye:~#

root@barreleye:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 02000000 00010000 "1e620000.fmc:flash@0"
mtd1: 00060000 00010000 "u-boot"
mtd2: 00020000 00010000 "u-boot-env"
mtd3: 00280000 00010000 "kernel"
mtd4: 001c0000 00010000 "initramfs"
mtd5: 01740000 00010000 "rofs"
mtd6: 00400000 00010000 "rwfs"
mtd7: 04000000 00001000 "1e630000.spi:pnor@0"

gkeishin, 2016-11-18:
- [x] Journal log  : 
[barreleye_v1.99.0-261-g730ecb8_journal.txt](https://github.com/openbmc/openbmc/files/599183/barreleye_v1.99.0-261-g730ecb8_journal.txt)

All the systems are in pretty bad shape when we come in to start the code update or test..

gkeishin, 2016-11-18:
As per Joel we need to ask Cedric why we got rid of the whole-of-flash mtd device

ratagupt, 2016-11-18:
Earlier the output of mtd is as follows.
mtd0: 02000000 00010000 "bmc"
mtd1: 00060000 00010000 "u-boot"
mtd2: 00020000 00010000 "u-boot-env"
mtd3: 00280000 00010000 "kernel"
mtd4: 001c0000 00010000 "initramfs"
mtd5: 01740000 00010000 "rofs"
mtd6: 00400000 00010000 "rwfs"
mtd7: 04000000 00001000 "pnor"

I had a chat with Cedric and he suggested that we cannot use labels under the flash device nodes in the device tree. This is allowed only under the partitions nodes.
He told that they removed that through the following commit..
https://github.com/openbmc/linux/commit/e62c7f11175525c028d9a6abaf6a2a5c275664d1.

gkeishin, 2016-11-18:
Thanks @ratagupt .. looking forward for the build with the revert changes to bring the systems up and operational with latter driver...

mine260309, 2016-11-18:
1. The update script `meta-phosphor/common/recipes-phosphor/initfs/files/obmc-update.sh` detects mtd name to decide which mtd device to flash. (See `findmtd()`)
2. The REST script `pyflashbmc/bmc_update.py` expect tarball, e.g. `romulus-20161117210452.all.tar`, and if extracted, the image file is `image-bmc`, so the above update script will try to find a mtd device with name `bmc`.

As I know that is how REST API works: it depends on the label name.
Now the `bmc` label is removed in kernel driver, so the update script fails to do update.

Why Cedric says `we cannot use labels under the flash device nodes in the device tree.`?

shenki, 2016-11-19:
There are two issues at play. One is the "bmc" label has gone from the mtd device that represents the entire flash.

Secondly, the mtd device that represents the entire flash has gone away. It's not clear to me if it has gone on both g4 and g5 systems, or just on g5.

@legoater what do you think?

legoater, 2016-11-19:
commit [e62c7f111755](https://github.com/openbmc/linux/commit/e62c7f11175525c028d9a6abaf6a2a5c275664d1) ("ARM: dts: aspeed: fix flash module device nodes") removed at the flash level on the label

* ```pnor``` on barreleye, firestone, garrison, palmetto, zaius
* ```alt```  on witherspoon, romulus
* ```bmc```  on all, as it came with the flash layout .dtsi file

@mine260309 : the changes were proposed to mainline and rejected because they were not following the Linux device tree guidelines.  The [jedec,spi-nor](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/mtd/jedec%2Cspi-nor.txt) device nodes should not contain a ```label``` property but the [partition](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/mtd/partition.txt) node can.

But this is not really practical to identify a specific flash module on a system with 3 or 4 ...  I will make some enquiry on how to address this need.

shenki, 2016-11-21:
@legoater I re-read upstream's comment on your proposed bindings and I think they are okay with adding a `label` property to the `jedec,spi-nor` nodes as long as we add that to the device tree bindings documentation.

legoater, 2016-11-21:
ok. I think it is indeed the best solution. I have restarted the thread on this particular topic to make sure my understanding is correct and I will send a patch re-adding these labels on the openbmc kernel.

legoater, 2016-11-21:
I spend some time today on this issue and it seems that the 'bmc' label disappearance is not due to the commit e62c7f111755. I will bisect tomorrow.

legoater, 2016-11-22:
My config was missing ```CONFIG_MTD_PARTITIONED_MASTER=y```. The interesting thing is that we still have this mtd device without the config option being set : 
```
mtd0: 02000000 00001000 "1e620000.fmc:flash@0"
```

But for a customize  label we need to set ```CONFIG_MTD_PARTITIONED_MASTER=y```

@shenki, @mine260309, shall I re-add the bmc labels  ```bmc``` and ```pnor``` or just ```bmc``` ? 

Thanks,

geissonator, 2016-11-28:
Hey guys, this one is a pretty serious hit to our automated testing, will a fix be coming soon?  We'd even take a quick hack if that's a better option for now.

shenki, 2016-11-28:
@legoater Re-add both please

shenki, 2016-11-28:
With @legoater's patch http://patchwork.ozlabs.org/patch/699881/

On Palmetto:

```
dev:    size   erasesize  name
mtd0: 02000000 00001000 "bmc"
mtd1: 00060000 00001000 "u-boot"
mtd2: 00020000 00001000 "u-boot-env"
mtd3: 00280000 00001000 "kernel"
mtd4: 001c0000 00001000 "initramfs"
mtd5: 01740000 00001000 "rofs"
mtd6: 00400000 00001000 "rwfs"
mtd7: 02000000 00010000 "pnor"
```

On Romulus:

```
# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 02000000 00001000 "bmc"
mtd1: 00060000 00001000 "u-boot"
mtd2: 00020000 00001000 "u-boot-env"
mtd3: 00280000 00001000 "kernel"
mtd4: 001c0000 00001000 "initramfs"
mtd5: 01740000 00001000 "rofs"
mtd6: 00400000 00001000 "rwfs"
mtd7: 02000000 00001000 "alt"
mtd8: 08000000 00001000 "pnor"
```

Looks good. I've applied it to dev-4.7 as of https://github.com/openbmc/linux/commit/b278d67c74167bd4089c3086a7a37ac54b8c86ae

shenki, 2016-11-28:
Please review: https://gerrit.openbmc-project.xyz/#/c/1308/

closed 2016-11-28 by williamspatrick

gkeishin, 2016-11-28:
This is good to go.. Tested and I see the MTD block BMC partition is back

`root@barreleye:~# cat /proc/mtd`
`dev:    size   erasesize  name`
`mtd0: 02000000 00010000 "bmc"`
`mtd1: 00060000 00010000 "u-boot"`
`mtd2: 00020000 00010000 "u-boot-env"`
`mtd3: 00280000 00010000 "kernel"`
`mtd4: 001c0000 00010000 "initramfs"`
`mtd5: 01740000 00010000 "rofs"`
`mtd6: 00400000 00010000 "rwfs"`
`mtd7: 04000000 00001000 "pnor"`

`root@barreleye:~# cat /etc/os-release`
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-296"
VERSION_ID="v1.99.0-296-ge743b5a-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-296"
BUILD_ID="v1.99.0"

`root@barreleye:~# obmcutil state`
` = BMC_READY`