openbmc/openbmc #817: can't fetch code from git://github.com/openbmc/linux

ceray0, 2016-11-18:
below is the bitbake output info,please help me check, where is wrong?thank you!

ysdeng@ubuntu:~/zmcheng/work/openbmc/build$ bitbake obmc-phosphor-image
Loading cache: 100% |####################################################################################| ETA:  00:00:00
Loaded 2487 entries from dependency cache.
Parsing recipes: 100% |##################################################################################| Time: 00:00:07
Parsing of 1923 .bb files complete (1922 cached, 1 parsed). 2488 targets, 374 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "i686-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-openbmc-linux-gnueabi"
MACHINE           = "palmetto"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta              
meta-poky         
meta-oe           
meta-networking   
meta-python       
meta-virtualization 
meta-phosphor     
meta-aspeed       
meta-ast2400      
meta-openpower    
meta-ibm          
meta-palmetto     = "master:b78c6def968a515c5b9f2d44f44c24b19134b6d5"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
$<50>WARNING: linux-obmc-4.7+gitAUTOINC+9e654d63fb-r0 do_fetch: Failed to fetch URL git://github.com/openbmc/linux;protocol=git;branch=dev-4.7, attempting MIRRORS if available
$<50>ERROR: linux-obmc-4.7+gitAUTOINC+9e654d63fb-r0 do_fetch: Fetcher failure: Fetch command failed with exit code 128, output:
Cloning into bare repository '/home/ysdeng/zmcheng/work/openbmc/build/downloads/git2/github.com.openbmc.linux'...
fatal: read error: Connection reset by peer

$<50>ERROR: linux-obmc-4.7+gitAUTOINC+9e654d63fb-r0 do_fetch: Function failed: Fetcher failure for URL: 'git://github.com/openbmc/linux;protocol=git;branch=dev-4.7'. Unable to fetch URL from any source.
$<50>ERROR: Logfile of failure stored in: /home/ysdeng/zmcheng/work/openbmc/build/tmp/work/palmetto-openbmc-linux-gnueabi/linux-obmc/4.7+gitAUTOINC+9e654d63fb-r0/temp/log.do_fetch.22772
$<50>ERROR: Task 52 (/home/ysdeng/zmcheng/work/openbmc/meta-phosphor/common/recipes-kernel/linux/linux-obmc_4.7.bb, do_fetch) failed with exit code '1'
$<50>NOTE: Tasks Summary: Attempted 458 tasks of which 439 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

Summary: 1 task failed:
  /home/ysdeng/zmcheng/work/openbmc/meta-phosphor/common/recipes-kernel/linux/linux-obmc_4.7.bb, do_fetch
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

mine260309, 2016-11-18:
Linux repo is huge, I guess it's due to network issues since the error says ` fatal: read error: Connection reset by peer`

amboar, 2017-08-11:
I think this should be closed - it's either a issue with Github infrastructure or end-user networking, not an OpenBMC bug (unless we consider our use of Github a bug).

gtmills, 2018-05-23:
Closing. If the issue comes back, please reopen.

closed 2018-05-23 by gtmills