openbmc/openbmc #820: Ensure builds are not -dirty

shenki, 2016-11-21:
Our CI builds come out as -dirty. This means git has found uncommitted artefacts in the build tree. We should resolve this, so we know when builds are actually dirty (ie, someone's been doing experiments on their workstation) or if it's a clean CI built image.

In addition to this, adding the Jenkins build info to the builds of master would be a great addition. Having this in `/etc/os-release` would allow builds to be traced back without having to match commits to builds.

shenki, 2016-11-21:
This could be a clue https://lists.yoctoproject.org/pipermail/yocto/2015-May/024832.html

```

It turns out "-dirty" has nothing directly to do with my own recipe at all.

It appears as soon as I specify 'KERNEL_PROVIDER = "linux-altera"' in local.conf, instead of allowing the meta-altera layer's default (which is "linux-altera-ltsi") to go through.
"linux-altera" currently pulls in kernel version 3.15, whereas "linux-altera-ltsi" pulls in 3.10.
```

Is the dirty tag due to the way we're using `local.conf`?

williamspatrick, 2017-02-28:
I figured this out.  

remote:   https://gerrit.openbmc-project.xyz/2750 os-release: ensure 'dirty' really means dirty.

closed 2017-03-01 by williamspatrick