openbmc/openbmc #821: [ Barreleye v1.99.0-271-g790e82a ]  BIOS versioning string changed ?

gkeishin, 2016-11-21:
The earlier driver BIOS version naming were in this format example ` open-power-barreleye-v1.8`
however in the new build we see this in this format of versioning `OpenPower-Barreleye-v1.0.1"`
so is this going to be the generic versioning ahead ?

` rango@ubuntu:~$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET https://xx.xx.xx.xx/org/openbmc/inventory/system/bios
`
{
  "data": {
    "Asset Tag": "", 
    "Custom Field 1": "\top-build-ead2390", 
    "Custom Field 2": "\tbuildroot-bc248fa", 
    "Custom Field 3": "\tskiboot-5.2.2", 
    "Custom Field 4": "\thostboot-dcf07f5-b88c8d8", 
    "Custom Field 5": "\tlinux-4.4.9-openpower2-919f476", 
    "Custom Field 6": "\tpetitboot-v1.1.1-307ab4d", 
    "Custom Field 7": "\tbarreleye-xml-ef8ad6a", 
    "Custom Field 8": "\tocc-19a3ebc", 
    "FRU File ID": "", 
    "Manufacturer": "", 
    "Model Number": "", 
    "Name": "OpenPOWER Firmware", 
    "Serial Number": "", 
    ``` "Version": "OpenPower-Barreleye-v1.0.1",  ``` 
    "fault": "False", 
    "fru_type": "SYSTEM", 
    "is_fru": 1, 
    "present": "True", 
    "version": ""
  }, 
  "message": "200 OK", 
  "status": "ok"
}rango@ubuntu:~$

shenki, 2016-11-21:
Those strings come from the VERSION partition in the PNOR. They are free-form strings that are generated by the op-build build process, so they may change at any time.

They are the output of a variant of `git describe`, which means the will include whatever git tags are present in op-build.

The op-build project may chose to strictly control the strings, but other vendors may not, so whatever code we write should be careful to accept different formats.

ratagupt, 2016-11-21:
seems we can make the change in the test code...Test code should be generic enough..

gkeishin, 2016-11-22:
I think we can do as much like making sure the Version Field is NOT empty and doesn't have wildcards character like $#%* etc... I that should do.. I don't think it does add any value nevertheless we can tweak the existing test code to do like this check..

@SrideviRamesh  Can u take a look at this ?

gkeishin, 2016-11-23:
Closing here.. will fix in Automation or we may decide to drop the changes if it doesn't add value after discussion. 

https://github.com/openbmc/openbmc-test-automation/issues/226

closed 2016-11-23 by gkeishin