openbmc/openbmc #823: Errors at startup on palmetto v1.99.0-265-g2d652fd-dirty

shenki, 2016-11-21:
```
Nov 21 14:37:00 palmetto systemd[1]: Started Wait for /org/openbmc/control/host0.
Nov 21 14:37:00 palmetto systemd[1]: org.openbmc.control.Power@0.service: Main process exited, code=dumped, status=11/SEGV
Nov 21 14:37:00 palmetto systemd[1]: Failed to start Phosphor Power0 Control.
Nov 21 14:37:00 palmetto systemd[1]: org.openbmc.control.Power@0.service: Unit entered failed state.
Nov 21 14:37:00 palmetto systemd[1]: org.openbmc.control.Power@0.service: Failed with result 'core-dump'.
Nov 21 14:37:00 palmetto control_host.exe[751]: GPIO Lookup:  FSI_ENABLE = 344,out
Nov 21 14:37:00 palmetto control_host.exe[751]: GPIO Lookup:  CRONUS_SEL = 326,out
Nov 21 14:37:00 palmetto control_host.exe[751]: GPIO Lookup:  BMC_THROTTLE = 395,out
Nov 21 14:37:00 palmetto control_host.exe[751]: GPIO Lookup:  IDBTN = 455,out
Nov 21 14:37:00 palmetto systemd[1]: Started OpenPOWER Host0 Control.
Nov 21 14:37:01 palmetto systemd[1]: org.openbmc.control.Power@0.service: Service hold-off time over, scheduling restart.
Nov 21 14:37:01 palmetto systemd[1]: Stopped Phosphor Power0 Control.
Nov 21 14:37:01 palmetto systemd[1]: Starting Phosphor Power0 Control...
Nov 21 14:37:01 palmetto systemd[1]: Starting Phosphor Settings Manager...
Nov 21 14:37:01 palmetto systemd[1]: org.openbmc.control.Power@0.service: Main process exited, code=dumped, status=11/SEGV
Nov 21 14:37:01 palmetto systemd[1]: Failed to start Phosphor Power0 Control.
Nov 21 14:37:01 palmetto systemd[1]: org.openbmc.control.Power@0.service: Unit entered failed state.
Nov 21 14:37:01 palmetto systemd[1]: org.openbmc.control.Power@0.service: Failed with result 'core-dump'.
Nov 21 14:37:02 palmetto systemd[1]: org.openbmc.control.Power@0.service: Service hold-off time over, scheduling restart.
Nov 21 14:37:02 palmetto systemd[1]: Stopped Phosphor Power0 Control.
Nov 21 14:37:02 palmetto systemd[1]: Starting Phosphor Power0 Control...
Nov 21 14:37:02 palmetto systemd[1]: org.openbmc.control.Power@0.service: Main process exited, code=dumped, status=11/SEGV
Nov 21 14:37:02 palmetto systemd[1]: Failed to start Phosphor Power0 Control.
Nov 21 14:37:02 palmetto systemd[1]: org.openbmc.control.Power@0.service: Unit entered failed state.
Nov 21 14:37:02 palmetto systemd[1]: org.openbmc.control.Power@0.service: Failed with result 'core-dump'.
Nov 21 14:37:03 palmetto systemd[1]: org.openbmc.control.Power@0.service: Service hold-off time over, scheduling restart.
Nov 21 14:37:03 palmetto systemd[1]: Stopped Phosphor Power0 Control.
Nov 21 14:37:03 palmetto systemd[1]: Starting Phosphor Power0 Control...
Nov 21 14:37:03 palmetto systemd[1]: org.openbmc.control.Power@0.service: Main process exited, code=dumped, status=11/SEGV
Nov 21 14:37:03 palmetto systemd[1]: Failed to start Phosphor Power0 Control.
Nov 21 14:37:03 palmetto systemd[1]: org.openbmc.control.Power@0.service: Unit entered failed state.
Nov 21 14:37:03 palmetto systemd[1]: org.openbmc.control.Power@0.service: Failed with result 'core-dump'.
Nov 21 14:37:04 palmetto systemd[1]: org.openbmc.control.Power@0.service: Service hold-off time over, scheduling restart.
Nov 21 14:37:04 palmetto systemd[1]: Stopped Phosphor Power0 Control.
Nov 21 14:37:04 palmetto systemd[1]: Starting Phosphor Power0 Control...
Nov 21 14:37:04 palmetto systemd[1]: org.openbmc.control.Power@0.service: Main process exited, code=dumped, status=11/SEGV
Nov 21 14:37:04 palmetto systemd[1]: Failed to start Phosphor Power0 Control.
Nov 21 14:37:04 palmetto systemd[1]: org.openbmc.control.Power@0.service: Unit entered failed state.
Nov 21 14:37:04 palmetto systemd[1]: org.openbmc.control.Power@0.service: Failed with result 'core-dump'.
Nov 21 14:37:05 palmetto systemd[1]: org.openbmc.control.Power@0.service: Service hold-off time over, scheduling restart.
Nov 21 14:37:05 palmetto systemd[1]: Stopped Phosphor Power0 Control.
Nov 21 14:37:05 palmetto systemd[1]: org.openbmc.control.Power@0.service: Start request repeated too quickly.
Nov 21 14:37:05 palmetto systemd[1]: Failed to start Phosphor Power0 Control.
Nov 21 14:37:06 palmetto systemd[1]: Started Phosphor Settings Manager.
Nov 21 14:37:06 palmetto systemd[1]: Starting Wait for /org/openbmc/settings/host0...
Nov 21 14:37:06 palmetto settings_manager.py[783]: Running HostSettingsService
Nov 21 14:37:08 palmetto systemd[1]: Started Wait for /org/openbmc/settings/host0.
Nov 21 14:37:08 palmetto systemd[1]: Reached target Restore Last State.
Nov 21 14:37:08 palmetto systemd[1]: Starting Restore Last State...
Nov 21 14:37:09 palmetto discover_system_state.py[812]: Traceback (most recent call last):
Nov 21 14:37:09 palmetto discover_system_state.py[812]:   File "/usr/sbin/discover_system_state.py", line 59, in <module>
Nov 21 14:37:09 palmetto discover_system_state.py[812]:     pgood = getProperty(bus, dbus_objects, 'power', 'pgood')
Nov 21 14:37:09 palmetto discover_system_state.py[812]:   File "/usr/sbin/discover_system_state.py", line 53, in getProperty
Nov 21 14:37:09 palmetto discover_system_state.py[812]:     objs[key]['bus_name'], objs[key]['object_name'], introspect=False)
Nov 21 14:37:09 palmetto discover_system_state.py[812]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
Nov 21 14:37:09 palmetto discover_system_state.py[812]:     follow_name_owner_changes=follow_name_owner_changes)
Nov 21 14:37:09 palmetto discover_system_state.py[812]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
Nov 21 14:37:09 palmetto discover_system_state.py[812]:     self._named_service = conn.activate_name_owner(bus_name)
Nov 21 14:37:09 palmetto discover_system_state.py[812]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
Nov 21 14:37:09 palmetto discover_system_state.py[812]:     self.start_service_by_name(bus_name)
Nov 21 14:37:09 palmetto discover_system_state.py[812]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
Nov 21 14:37:09 palmetto discover_system_state.py[812]:     'su', (bus_name, flags)))
Nov 21 14:37:09 palmetto discover_system_state.py[812]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Nov 21 14:37:09 palmetto discover_system_state.py[812]:     message, timeout)
Nov 21 14:37:09 palmetto discover_system_state.py[812]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.control.Power was not provided by any .service files
Nov 21 14:37:09 palmetto systemd[1]: obmc-mgr-state.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 14:37:09 palmetto systemd[1]: Failed to start Restore Last State.
Nov 21 14:37:09 palmetto systemd[1]: obmc-mgr-state.service: Unit entered failed state.
Nov 21 14:37:09 palmetto systemd[1]: obmc-mgr-state.service: Failed with result 'exit-code'.
Nov 21 14:37:09 palmetto systemd[1]: Reached target Phosphor System - Standby.
```

ratagupt, 2016-11-21:
ov 21 14:37:00 palmetto systemd[1]: Started Wait for /org/openbmc/control/host0.
Nov 21 14:37:00 palmetto systemd[1]: org.openbmc.control.Power@0.service: Main process exited, code=dumped, status=11/SEGV
Nov 21 14:37:00 palmetto systemd[1]: Failed to start Phosphor Power0 Control.
Nov 21 14:37:00 palmetto systemd[1]: org.openbmc.control.Power@0.service: Unit entered failed state.
Nov 21 14:37:00 palmetto systemd[1]: org.openbmc.control.Power@0.service: Failed with result 'core-dump'.

Seems that power application got core dumped while it was being started...and discovery system sate threw the exception as power application was not started...

shenki, 2016-11-21:
Also present in v1.99.0-264-g6bd26a3-dirty (6bd26a3)

ratagupt, 2016-11-21:
is it related to this one?
https://github.com/openbmc/skeleton/commit/0475f65f5feb9381ef2710dca96b545546ccd944...
@adamliyi :- Can you take a quick look on this?

shenki, 2016-11-21:
```
6bd26a357e2db50999a2ee07537bdacb3ad841de is the first bad commit
commit 6bd26a357e2db50999a2ee07537bdacb3ad841de
Author: Patrick Williams <patrick@stwcx.xyz>
Date:   Fri Nov 18 09:10:02 2016 -0600

    skeleton-rev.bbclass: bump version
    
    Change-Id: Ib076b153143cca72a2d2a48b93325f4c4b40eb62
    Signed-off-by: Patrick Williams <patrick@stwcx.xyz>
```

As @ratagupt says, the first bad commit is the change in skeleton.

mine260309, 2016-11-21:
My wild guess: it could be the same root cause as #818, I was meeting the same issue last week.
Suggest pick the latest skeleton (d560aac) and see if it works.

shenki, 2016-11-22:
This is fixed with @mine260309 's patch, which is included as of [v1.99.0-275-g63248b4](63248b4)

closed 2016-11-22 by shenki