openbmc/openbmc #824: [ Barreleye v1.99.0-271-g790e82a ] Poweron didnt boot to HOST_BOOTED

gkeishin, 2016-11-21:
The newer driver build power on failed .. v1.99.0-271-g790e82a

- [x] Journal log 
[barreleye_v1.99.0-271-g790e82a_journal.txt](https://github.com/openbmc/openbmc/files/602846/barreleye_v1.99.0-271-g790e82a_journal.txt)

As per @geissonator input

This new issue may be related to the PCIe workaround we brought in, I see this during power on:

```
Oct 20 20:00:14 barreleye settings_manager.py[829]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Oct 20 20:00:14 barreleye settings_manager.py[829]: Traceback (most recent call last):
Oct 20 20:00:14 barreleye settings_manager.py[829]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Oct 20 20:00:15 barreleye settings_manager.py[829]:     self._handler(*args, **kwargs)
Oct 20 20:00:15 barreleye settings_manager.py[829]: TypeError: settings_signal_handler() takes exactly 5 arguments (4 given)
Oct 20 20:00:15 barreleye pcie_slot_present.exe[880]: Checking Presence: SLOT0_RISER_PRESENT
```

adamliyi, 2016-11-21:
If it is related the PCIe workaround from my patch - I will have a look.

adamliyi, 2016-11-22:
The error: `TypeError: settings_signal_handler() takes exactly 5 arguments (4 given)` looks to be caused by commit: https://github.com/openbmc/phosphor-settingsd/commit/62b55f35fde5e057bb7b0294f2e2d8ad66170e2d#diff-3cc9423a40eac5e80076950f02247305R150
@SS-IBM , I've put a fix at: https://gerrit.openbmc-project.xyz/#/c/1257/. Please have a look.

closed 2016-11-22 by williamspatrick