openbmc/openbmc #825: discover_system_state: Exception at boot

shenki, 2016-11-21:
First boot after flashing v1.99.0-264-g6bd26a3-dirty (6bd26a3) on a palmetto

```
Nov 21 15:06:06 palmetto systemd[1]: Started Wait for /org/openbmc/settings/host0.
Nov 21 15:06:06 palmetto systemd[1]: Reached target Restore Last State.
Nov 21 15:06:06 palmetto systemd[1]: Starting Restore Last State...
Nov 21 15:06:07 palmetto discover_system_state.py[821]: Traceback (most recent call last):
Nov 21 15:06:07 palmetto discover_system_state.py[821]:   File "/usr/sbin/discover_system_state.py", line 59, in <module>
Nov 21 15:06:07 palmetto discover_system_state.py[821]:     pgood = getProperty(bus, dbus_objects, 'power', 'pgood')
Nov 21 15:06:07 palmetto discover_system_state.py[821]:   File "/usr/sbin/discover_system_state.py", line 53, in getProperty
Nov 21 15:06:07 palmetto discover_system_state.py[821]:     objs[key]['bus_name'], objs[key]['object_name'], introspect=False)
Nov 21 15:06:07 palmetto discover_system_state.py[821]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
Nov 21 15:06:07 palmetto discover_system_state.py[821]:     follow_name_owner_changes=follow_name_owner_changes)
Nov 21 15:06:07 palmetto discover_system_state.py[821]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
Nov 21 15:06:07 palmetto discover_system_state.py[821]:     self._named_service = conn.activate_name_owner(bus_name)
Nov 21 15:06:07 palmetto discover_system_state.py[821]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
Nov 21 15:06:07 palmetto discover_system_state.py[821]:     self.start_service_by_name(bus_name)
Nov 21 15:06:07 palmetto discover_system_state.py[821]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
Nov 21 15:06:07 palmetto discover_system_state.py[821]:     'su', (bus_name, flags)))
Nov 21 15:06:07 palmetto discover_system_state.py[821]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Nov 21 15:06:07 palmetto discover_system_state.py[821]:     message, timeout)
Nov 21 15:06:07 palmetto discover_system_state.py[821]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.control.Power was not provided by any .service files
Nov 21 15:06:07 palmetto systemd[1]: obmc-mgr-state.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 15:06:07 palmetto systemd[1]: Failed to start Restore Last State.
Nov 21 15:06:07 palmetto systemd[1]: obmc-mgr-state.service: Unit entered failed state.
Nov 21 15:06:07 palmetto systemd[1]: obmc-mgr-state.service: Failed with result 'exit-code'.
Nov 21 15:06:07 palmetto systemd[1]: Reached target Phosphor System - Standby.
```

shenki, 2016-11-21:
Same issue as https://github.com/openbmc/openbmc/issues/823

mine260309, 2016-11-21:
I have a local build on latest master with #818 fixed, the issue mentioned in #823 is fixed, there're no more power_control.exe crash, but this issue related to "discover_system_state.py" remains.
So I'd say they are different issues.

`obmc-mgr-state.service` seems depends on other services, e.g. power control, but `obmc-mgr-state.service` seems not defines such dependency.
I'm not very familiar with systemd yet, @williamspatrick could you check if `obmc-mgr-state.service` was incorrectly configured?

shenki, 2016-11-22:
I'm still seeing this with [v1.99.0-275-g63248b4-dirty](63248b4).

closed 2019-04-27 by stale[bot]