openbmc/openbmc #826: Optimize or fix the 20 minute boot watchdog timeout

geissonator, 2016-11-22:


shenki, 2016-11-22:
I thought about contacting upstream systemd to ask if they would add a way to extend the watchdog timeout for that one shutdown. I suspect their answer would be to not do work in the shutdown path that is expected to take a long time.

Instead we should put the system in a state that is like shutdown, where other userspace processes are not running, and the filesystems are all unmounted, to do our upgrade. Once the upgrade is complete we should then shutdown.

amboar, 2016-11-22:
@shenki what you describe is effectively what it is: Systemd has transitioned us out of the root filesystem back into initramfs and unmounted everything ready for us to upgrade. The problem is the means to get there is to invoke a shutdown/reboot, and by that point the watchdog is already set. I think systemd has exec'ed to our process to get out of its own way thus it can't reconfigure the watchdog afterwards.

Maybe we could add a helper that does?

https://www.freedesktop.org/wiki/Software/systemd/InitrdInterface/

geissonator, 2017-10-13:
Gerrit is down but I thought we fixed this

geissonator, 2018-06-04:
@eddiejames - I think with https://gerrit.openbmc-project.xyz/#/c/8098/5/meta-phosphor/common/recipes-core/systemd/systemd/0002-core-Add-WatchdogDevice-config-option-and-implement-.patch we went back to the default 10 minutes for this?

mdmillerii, 2018-06-04:
~~I thought I~~ pushed a patch to start and stop the watchdog daemon in busybox during the call to update from the shutdown script, setting a new timeout?

Ahh yes, here:  It says it was merged, but I don't see the result in the current git tree, and I am not finding the commit that caused this not to occur.

Ahh here it is: 589ebdf3abb10ee55a0acb7364d9a0d8d434dbc5

Hmm, was it only merged into 1.99x and not the master branch?

Can someone please fix this?

amboar, 2018-06-04:
> Hmm, was it only merged into 1.99x and not the master branch?

How do you come to that conclusion @mdmillerii ?

```
$ git branch -r --contains 589ebdf3abb10ee55a0acb7364d9a0d8d434dbc5 
  gerrit/master
  origin/HEAD -> origin/master
  origin/master
```
Or have we reverted/reworked the code?

mdmillerii, 2018-06-08:
I was diffing the contents of that commit with the current contents (renamed file) and saw two differences (this commit dropped and copies changed from 1 to 2), but I was not finding any commits when doing `git log <refspec> -- <old-name> <new-name>`.  Perhaps that syntax is wrong, I haven't had a chance to go back and figure out the true changes here.

mdmillerii, 2018-06-12:
Looking further, it appears to have been merged after the rename and is in the current master, so this issue should be closed as fixed by 589ebdf3abb10ee55a0acb7364d9a0d8d434dbc5 (and as noted the revert to the 20min timeout was 24e0fbea9bf1b05155c8dc9776a0bf4d5d45b15f and reverted in #1938.

geissonator, 2018-06-18:
Closing per Miltons comments above.

closed 2018-06-18 by geissonator