openbmc/openbmc #827: Exception when powering off v1.99.0-275-g63248b4-dirty

shenki, 2016-11-22:
On a system that is also showing #825, the system comes up in `HOST_BOOTED` state. Doing a poweroff results in this:

```
Nov 22 02:41:39 palmetto chassis_control.py[684]: Turn off power
Nov 22 02:41:39 palmetto systemd[1]: Starting Wait for /org/openbmc/watchdog/host0...
Nov 22 02:41:39 palmetto systemd[1]: Created slice system-obmc\x2dstop\x2dwatchdog.slice.
Nov 22 02:41:39 palmetto systemd[1]: Created slice system-op\x2dpower\x2dstop.slice.
Nov 22 02:41:39 palmetto systemd[1]: Started Stop Power0.
Nov 22 02:41:39 palmetto systemd[1]: Created slice system-op\x2dwait\x2dpower\x2doff.slice.
Nov 22 02:41:39 palmetto systemd[1]: Starting Wait for Power0 to turn off...
Nov 22 02:41:40 palmetto host_watchdog.exe[671]: Stopping watchdog
Nov 22 02:41:40 palmetto systemd[1]: Started Wait for /org/openbmc/watchdog/host0.
Nov 22 02:41:40 palmetto system_manager.py[716]: Running System State: HOST_POWERING_OFF
Nov 22 02:41:40 palmetto power_control.exe[754]: PowerControl: setting power up POWER_PIN to 1
Nov 22 02:41:40 palmetto systemd[1]: Starting Stop Watchdog0...
Nov 22 02:41:40 palmetto systemd[1]: Started Stop Watchdog0.
Nov 22 02:41:41 palmetto power_control.exe[754]: PowerControl: setting reset USB_RESET to 1
Nov 22 02:41:41 palmetto power_control.exe[754]: PowerControl: setting pci reset PCIE_RESET to 1
Nov 22 02:41:41 palmetto system_manager.py[716]: Running System State: HOST_POWERED_OFF
Nov 22 02:41:41 palmetto sensor_manager2.py[685]: Deleting OCC device
Nov 22 02:41:41 palmetto settings_manager.py[769]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Nov 22 02:41:41 palmetto settings_manager.py[769]: Traceback (most recent call last):
Nov 22 02:41:41 palmetto settings_manager.py[769]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Nov 22 02:41:41 palmetto settings_manager.py[769]:     self._handler(*args, **kwargs)
Nov 22 02:41:41 palmetto settings_manager.py[769]: TypeError: settings_signal_handler() takes exactly 5 arguments (4 given)
Nov 22 02:41:41 palmetto systemd[1]: Started Wait for Power0 to turn off.
Nov 22 02:41:41 palmetto systemd[1]: Reached target Power0 Off.
Nov 22 02:41:41 palmetto systemd[1]: Reached target Chassis0 (Stop).
```

ratagupt, 2016-11-22:
https://github.com/openbmc/openbmc/issues/824.

Seems it is same...

ratagupt, 2016-11-22:
@adamliyi :- Can you take a look is this same as 824?

geissonator, 2016-12-01:
Yes, this looks like a duplicate of #824 with the 5 vs 4 argument issue.

closed 2016-12-01 by geissonator