openbmc/openbmc #828: [Barreleye v1.99.0-280-g29af197] Poweron broken... Failed to boot HOST

gkeishin, 2016-11-22:
Powering on stuck at HOST_POWERING_ON

- [x] Journal log   
[barreleye_v1.99.0-280-g29af197_poweron_journal.txt](https://github.com/openbmc/openbmc/files/606685/barreleye_v1.99.0-280-g29af197_poweron_journal.txt)

```
Oct 20 20:12:53 barreleye systemd[1]: org.openbmc.ObjectMapper.service: Start operation timed out. Terminating.
Oct 20 20:12:53 barreleye systemd[1]: Failed to start Phosphor DBus Service Discovery Manager.
Oct 20 20:12:53 barreleye systemd[1]: org.openbmc.ObjectMapper.service: Unit entered failed state.
Oct 20 20:12:53 barreleye systemd[1]: org.openbmc.ObjectMapper.service: Failed with result 'timeout'.
Oct 20 20:12:54 barreleye systemd[1]: org.openbmc.ObjectMapper.service: Service hold-off time over, scheduling restart.
Oct 20 20:12:54 barreleye systemd[1]: Stopped Phosphor DBus Service Discovery Manager.
```

ratagupt, 2016-11-22:
Oct 20 19:58:20 barreleye phosphor-mapper[694]: ObjectMapper discovery complete
Oct 20 19:59:17 barreleye kernel: random: nonblocking pool is initialized
Oct 20 19:59:17 barreleye kernel[670]: random: nonblocking pool is initialized
Oct 20 20:02:45 barreleye systemd[1]: org.openbmc.ObjectMapper.service: Start operation timed out. Terminating.
Oct 20 20:02:45 barreleye systemd[1]: Failed to start Phosphor DBus Service Discovery Manager.
Oct 20 20:02:45 barreleye systemd[1]: org.openbmc.ObjectMapper.service: Unit entered failed state.
Oct 20 20:02:45 barreleye systemd[1]: org.openbmc.ObjectMapper.service: Failed with result 'timeout'.

Mapper.service didn't start..thats why subsequent failures...

ratagupt, 2016-11-22:
Oct 20 20:02:45 barreleye mapper[731]: Error starting event loop: No route to host
Oct 20 20:02:46 barreleye mapper[735]: Error starting event loop: No route to host
 phosphor-objmgr/libmapper/app.c 

	r = sd_event_loop(loop);
	if(r < 0) {
		fprintf(stderr, "Error starting event loop: %s\n",
				strerror(-r));
		goto finish;
}
seems while running the event loop is getting failed in the mapper..

ratagupt, 2016-11-22:
@bradbishop :- Can you please help?

adamliyi, 2016-11-22:
@ratagupt , I did some quick investigate, the cause may be the avahi daemon
```
Oct 20 21:20:35 barreleye phosphor-mapper[1247]: ObjectMapper startup complete.  Discovery in progress...
Oct 20 21:20:36 barreleye avahi-daemon[690]: ../../avahi-0.6.32/avahi-daemon/chroot.c: open() failed: No such file or directory
Oct 20 21:20:36 barreleye avahi-daemon[681]: Failed to open /usr/share/dbus-1/interfaces/org.freedesktop.Avahi.Server.xml: Invalid argument
Oct 20 21:20:36 barreleye avahi-daemon[681]: Failed to load introspection data.
Oct 20 21:20:46 barreleye phosphor-mapper[1247]: ObjectMapper discovery complete
```

ratagupt, 2016-11-22:
@adamliyi :- How it could be avahi-daemon?Not able to understand...Why the mapper sd_event_loop will fail due to avahi?

ratagupt, 2016-11-23:
Mapper discovery started at
 Oct 20 19:58:09 barreleye phosphor-mapper[694]: ObjectMapper startup complete.  Discovery in progress...
Completed at
Oct 20 19:58:20 barreleye phosphor-mapper[694]: ObjectMapper discovery complete >>>>>completed
Oct 20 19:59:17 barreleye kernel: random: nonblocking pool is initialized
Oct 20 19:59:17 barreleye kernel[670]: random: nonblocking pool is initialized
Oct 20 20:02:45 barreleye systemd[1]: org.openbmc.ObjectMapper.service: Start operation timed out. Terminating.

Oct 20 20:02:46 barreleye mapper[736]: Error starting event loop: No route to host
Oct 20 20:02:46 barreleye mapper[737]: Error starting event loop: No route to host

bradbishop, 2016-11-23:
This is fixed with https://gerrit.openbmc-project.xyz/#/c/981/

There was a small window yesterday where master was broken...we seem to have grabbed a build that was created during that window.

closed 2016-11-23 by bradbishop