openbmc/openbmc #829: [Barreleye/Palmetto] Non-SSL Connection to port 443 Error

gkeishin, 2016-11-22:
This failure has been in automation for quite sometime. We had a discussion on this with @bradbishop since it was not really a priority but now its a consistent failure and we would like to close on this..

**Test case expectation**

`This testcase is for test to check OpenBMC machine will not accepts the non-secure connection that is with http to port 443 and expect 400 in response`

**In actual Test result**

`Start / End / Elapsed: 20160913 22:22:25.494 / 20160913 22:22:29.659 / 00:00:04.165
22:22:29.658 FAIL ConnectionError: HTTPConnectionPool(host='9.3.29.119', port=443): Max retries exceeded with url: //list (Caused by ProtocolError('Connection aborted.', error(104, 'Connection reset by peer')))`

``` Earlier : We use to get HTTP_BAD_REQUEST for non-SSL Connection to port 443 ```

The test looks like this

```
Test non-SSL Connection to port 443
    [Documentation]     This testcase is for test to check OpenBMC machine
    ...     will not accepts the non-secure connection that is with http to
    ...     port 443 and expect 400 in response
    Create Session    openbmc    http://${OPENBMC_HOST}:443/
    ${resp}=    Get Request    openbmc   /list
    Should Be Equal As Strings    ${resp.status_code}    ${HTTP_BAD_REQUEST}
    Should Be Equal     ${resp.content}     Bad Request
```

Let us know if there is a change in the REST server response for this..

ratagupt, 2016-11-23:
I think that HTTP ERROR 4xx is for (request data is not valid).Now as in this scenario we are sending the http request to port 443,as the ssl connection would not be established so server doesn't get a chance to look at the request.so you should get error 104. which is connection reset by peer.

gkeishin, 2016-11-23:
@bradbishop  Can you confirm on @ratagupt statement above. Thanks

gkeishin, 2017-01-18:
Closing this .. Test now checks for the connection error

closed 2017-01-18 by gkeishin